xoseperez/espurna #1499: Cheap FR (EU 3 pin) plug with power consumption monitoring recommendation?

evlo, 2019-01-19:
Hi,

can you recommend cheap EU 3pin/FR plug with power monitoring, compatible with home assistant? Ideally as it is, but I can deal with flashing ESP firmware on it. Or I guess support any open standard that is.

This is popular ODM plug that is working with ESP firmwares
https://www.aliexpress.com/item/BlitzWolf-BW-SHP2-WIFI-Smart-Socket-EU-Plug-220V-16A-Remote-Control-Smart-Timing-Switch-Work/32871562977.html but it is EU 2pin

for my needs I could even use the led color control on this
https://www.aliexpress.com/item/BlitzWolf-BW-SHP6-EU-Plug-Metering-Version-WIFI-Smart-Socket-220V-240V-10A-Work-with-Amazon/32933098972.html?spm=2114.10010108.1000013.1.b1f579bcZ4dCtT&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.90158.0&scm_id=1007.13339.90158.0&scm-url=1007.13339.90158.0&pvid=b629b95f-af90-468c-b984-f616281d69f1 - but in lot of the plugs power monitoring is not working when plugged to 3pin instead of 2pin socket. Also on the second one i'm unsure about open standards support or even if ESP is possible.

Can you recommend some? Ideally this small style. Ideally 16A.

xoseperez, 2019-01-20:
The SHP6 uses the same pinout as the SHP2, so it's also supported by ESPurna (see #1429, I have a couple of them at home).

evlo, 2019-01-20:
Ok, but still 10A only. Weirdly I did found only this other one for FR/3 pin system https://www.aliexpress.com/store/product/Lonsonho-2-In-1-Smart-Plug-Wifi-Smart-Socket-EU-KR-Plug-Outlet-Power-Monitor-Energy/4045016_32952681576.html?spm=2114.10010108.1000023.1.e33354ebPa6HB7

Weird that 3 pin system is so much limiting.

Then there are some "brand names" from tp-link etc. But I was looking for alternative to sonoff pow2 that would be just plug and play. 
Sonoff's advantage is, that it can work with home assistant without flashing espurna.

BTW i tried google home and was surprised that power consumption measuring does not work in it at all.

closed 2019-03-28 by stale[bot]