xoseperez/espurna #1501: Sonoff LED: Set ColorTemperature and Brightness over MQTT

jbaumann, 2019-01-20:
Sonoff LED contains a warm and cold white led strip which can currently be controlled independently via MQTT. It would be nice to have an integrated control method that allows to change brightness (works already using the web interface) and the color temperature instead of fiddling with the different channels.

copyrights, 2019-05-20:
This would be useful for QuinLED boards, too.

copyrights, 2019-05-20:
I just had a look. while brightness is already partially working (see #1292), color temperature will be a larger change. Current implementation uses RGB leds. While we use two types of white leds. Mine are warm white (approx. 2700-3000 °K) and cold white (approx. 6000 °K).

PR #1732 (2019-05-26): Two channel CCT
This adds CCT (warm white / cold white) support to two light channel devices.

Together with #1730  this shall close #1501.
closed 2019-05-26 by mcspr