xoseperez/espurna #1502: Digoo NX-SP202 smart wifi socket support possible?

kobuki, 2019-01-20:
I've purchased a "Digoo NX-SP202" dual smart socket with power metering capability which is most probably using a ESP8266 module variant. I've made a few photos and there are a few interesting chips on the board. I didn't take it apart more as I was afraid of breaking it. If hints of support are positive, I'll take an attempt at complete disassembly. In general, I'm reluctant in giving out my WiFi password to a cloud-only device which controls any mains outlet so I bought it with the anticipation of being able to apply a 3rd-party OSS firmware.

Chips are (first image, from left to right):
1. KP3210SG / AK7EPG.1 / H38 - Kiwi Instruments switching regulator controller
2. BL 0937 / 1818AYH / D798090F - Shanghai Belling "Single phase, built-in oscillator, active power, RMS, and energy metering" IC
3. AMS1117 / 3.3 DN17 - 3.3V LDO

For the metering IC, there's a Chinese PDF available. Based on the images in the PDF it uses a simple serial communication protocol on 3.3V.

[image]

On the second image, at the top, the WiFi board is clearly visible. It reads KZL827VA and a (manufacturing?) date. Looks very much like the other WiFi boards seen in other espurna compatible devices.

[image]

Outside image:

[image]

kobuki, 2019-01-20:
Upon further investigation, it looks like a "Blitzwolf" variant with 2 relays and utilizing a single BL0937 which already has support. It requires desoldering the plug pins from the board (a very stupid but somewhat sturdy choice) so I'll leave it for later but I think it's supportable.

ColinShorts, 2019-01-20:
I think you'll need to disassemble further to identify the vertical module, and maybe see if there are any easily accessible solder points for flashing. I don't think it'll use boards anything like their IP cameras (Linux based), so it'll hopefully be esp based. The BL0937 is supported, so seems plausible.

kobuki, 2019-01-20:
Yeah, I plan to fully disassemble it later. The MAC address starts with ce:50:e3 which is a locally administered one, according to the RFC. They're not using an official MAC allotted for Espressif. Fishy, though it works.

ColinShorts, 2019-01-20:
I thought that address looked familiar, but it's not in my collection (from dhcpd.conf):

class "iot1" { match if substring (hardware,1,3) = 18:fe:34; }
class "iot2" { match if substring (hardware,1,3) = 2c:3a:e8; }
class "iot3" { match if substring (hardware,1,3) = 5c:cf:7f; }
class "iot4" { match if substring (hardware,1,3) = 60:01:94; }
class "iot5" { match if substring (hardware,1,3) = a0:20:a6; }
class "iot6" { match if substring (hardware,1,3) = dc:4f:22; }
class "iot7" { match if substring (hardware,1,3) = 84:0d:8e; }

It might be used on one of the splashproof sockets I bought from AliExpress that I've never used yet, or some bulbs that got too hot to use in an enclosed fitting... Nothing in the logs either. It'll bug me...

Anyways, if it's based on a blitzwolf you'll likely be in luck.

xoseperez, 2019-01-20:
I also have one of these partially disassembled and I'm facing the same problem. I have not yet dared to desolder the plug pins, but it looks like the only way to reach the uC...

xoseperez, 2019-01-20:
By the way, mine comes under the brand Oxaoxe, same model NX-SP202. Amazon [ES]: https://www.amazon.es/gp/product/B07G2JGYND

kobuki, 2019-01-20:
Yeah, it looks the same model under different branding on the outside. I think I'll take the plunge tomorrow and desolder the pins.

kobuki, 2019-01-21:
I made an attempt at desoldering the pins but I failed. I have little experience removing solder from such a mass. It's very hard to heat it to liquify the solder and then it's a very narrow place to work in. If anyone has suggestions on desoldering the pins (see the 2 large blobs of solder in my images), please reply.

kobuki, 2019-01-21:
Some further investigation revealed a few pins ([see the top of this image](https://user-images.githubusercontent.com/6766047/51495893-31eb3280-1dbe-11e9-94cf-c4419d42d98f.jpg)). The wifi board has 2x7 pins of which the leftmost 2 aren't connected to the main board but present.

Top row, pin 6 - Vcc
Bottom row:
pin 1 - right button (same image, on the other side of the board)
pin 6 - left button

That's what simple DMM probing has shown. I don't know about the rest but since the number of pins is the same, the Vcc/GND is in the same position, I think for now it's possible to assume the layout is the same as for a Blitzwolf type socket so a reflash is likely possible.

If it indeed can be reflashed, OTA would be available for later updates, right?

gn0st1c, 2019-01-22:
pin layout:
(shielded part pins = LEFT, bottom pins = RIGHT)

[image]

kobuki, 2019-01-22:
@gn0st1c: Thanks. Is it from the same model or some module from another smart socket?

If I reflash it, OTA would be available for later updates, right?

xoseperez, 2019-01-22:
Yes, you will be able to flash it OTA after.
Now we have to identify the GPIO for the two relays, the 3 GPIOs for the BL0937 (SEL, CF and CF1), the two LED and the button.
What temperature did you use to remove the blob of solder?

kobuki, 2019-01-22:
Actually it has 2 buttons, 2 2-color (red/blue) LEDs, the BL0937 and the 2 relays. A lot of IO is needed.

I used 350°C and a solder suction tool. There're 3 blobs to take care of: line, null and ground. The latter is a flat pin between the relays.

kobuki, 2019-01-25:
Does anyone have more info on the KP3210SG high voltage buck converter IC? I don't want to fiddle with this board plugged in the mains and I don't have an isolation transformer.  My hunch is the IC can take lower voltages like 24VAC and that would make further testing possible. Sadly I haven't found anything more than [this pdf](http://www.kiwiinst.com/ashx/DownloadFile.ashx?upfiles=mtscs/4d9cdfc7-ad0c-41b1-9a86-cf3bf6516068/20181019132527123.pdf) with basic information, it's more like the intro page of the full specs.

kobuki, 2019-02-21:
@xoseperez: since the support of socket has a deadline in 1.13.5, does that mean there's some progress in hacking it? Is there any more info on that?

xoseperez, 2019-02-21:
I will first close 1.13.4 and then start working on the new device...

evaluationevaluation, 2019-02-24:
did anyone find out the GPIOs for BL0937? I just managed to flash Slitinto NX-SP202 OTA using tuya-convert and now searching for the right template values... assume that the layout is similar/equal.

see also [https://github.com/arendst/Sonoff-Tasmota/issues/3950] for OTA flashing of TUYA-based devices.

evaluationevaluation, 2019-02-24:
perhaps found it out by my own with Trial&Error
here is my current configuration for Slitinto NX-SP202
Perhaps someone can doublecheck it and eventually checks this into GIT as a new device?

GPIO0: Button1 (17)
GPIO4: BL0937 CF (134)
GPIO5: HLWBL CF1 (132)
GPIO13: HLWBL SEL (130)
GPIO14: Relay2 (22)
GPIO15: Relay1 (21)
GPIO16: Button2 (18)

Had to adjust Voltage with Command:
VoltageSet 233
//233V = Voltage measured with a multimeter
Current needs also adjustment but for this I have to find a good calibration device (i.e. good old light bulb)
then you can adjust Current with command CurrentSet xxx (xxx in mA)
Wattage value displayed was plausible out of the box.

kobuki, 2019-02-24:
Nice. My NX-SP202 has 2 programmable LEDs, too, did you find the GPIO for those?

kobuki, 2019-02-24:
It seems the [Tasmota](https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations) guys are taking care of this device too, also see image. The Vcc/GND pins are matching my findings.

[image]

evaluationevaluation, 2019-02-25:
Have to correct myself. My configuration was not completely correct according the metering functionality.
After long trial&error I finally found the device Digoo DG-SP202 already in the sonoff_template.h and this configuration works like a charm with my Slitinto NX-SP202.
Now metering shows plausible values, too. :-)

kobuki, 2019-02-25:
@evaluationevaluation: but you're actually using Sonoff-Tasmota, right? Since they do support the device since a recent commit. I think info exchange is good between teams of OSS, but this is the ESPurna issue list, and ESPurna doesn't support it yet.

evaluationevaluation, 2019-02-25:
oh, sorry about the confusion.

xoseperez, 2019-03-04:
Finally flashed it using [tuya-convert project](https://github.com/ct-Open-Source/tuya-convert) with no problems. Only changed the current, voltage and power ratios to match those I use with testing resistive loads.

kobuki, 2019-03-04:
@xoseperez: great news! I haven't used tuya-convert yet, the small army of various firmware (yours, tasmoga, espeasy, etc.) are pretty new to me. It's capable of flashing alternate firmware to tuya-originated devices without opening them - do I get that right? So it's best I solder everything back in place and do the reflash using tuya-convert via wifi instead?

xoseperez, 2019-03-04:
Yes, so far I have been able to flash all the devices based on Tuya modules: the Oxaoxe NX-SP202 (also branded Digoo), the Blitzwolf BW-SHP6 and the GBLife RGBW Socket. Best way is to do it from a Raspberry Pi following the instructions in the README of the tuya-converter, project. Then I just upload the custom ESPurna image t the device:

```
curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/espurna-1.13.5-gblife-rgbw-socket.bin
```

kobuki, 2019-03-04:
Thanks. Looking at the tuya-convert repo it doesn't seem to require anything specific to the Pi series, so I'll attempt it from a Debian VM first.

kobuki, 2019-03-06:
I reflashed my device using tuya-convert. I'm using tasmota for now. I'll try espurna later. Just 2 questions. How are the following constants determined, manual calibration? How accurate is the meter chip?

```
#define HLW8012_CURRENT_RATIO       23296
#define HLW8012_VOLTAGE_RATIO       310085
#define HLW8012_POWER_RATIO         3368471
```

xoseperez, 2019-03-17:
I use manual procedure to define those constants using two different restive loads (an incandescent 60W light bulbs and a 900W heater) and using a (cheap) current clamp as a reference. But I don't have a systematic procedure to define how accurate the readings are.

closed 2019-03-25 by xoseperez