xoseperez/espurna #1503: MQTT: If status is sent as JSON in {root}/data, the retain config is ignored

jbaumann, 2019-01-20:
If I choose "Use JSON payload" in the MQTT config, then the configuration of the Retain flag is ignored and the status is always sent without the Retain flag. If I choose to send the status to different topics these are sent with a Retain flag when so configured.

Version 1.13.3, device Sonoff_LED.

mcspr, 2019-01-21:
https://github.com/xoseperez/espurna/blob/e88a860511b703740ce45f1d5e84dfc5c2d81d31/code/espurna/mqtt.ino#L639
does not use _mqtt_retain setting, unlike plaintext methods. wonder if this is intended because of scattered nature of json messaging

xoseperez, 2019-02-09:
It is intended since you cannot relay on what contents it will report. The whole approach of the JSON payload should be refactored.

closed 2019-02-09 by xoseperez