xoseperez/espurna #1504: Generic Relay v4.0 active low

filipemurias, 2019-01-20:
I have bought a ESP-01S relay module **v4.0** (https://www.aliexpress.com/item/ESP8266-ESP-01S-5V-WiFi-Relay-Module-Things-Smart-Home-Remote-Control-Switch-for-Arduino-Phone/32890526507.html?spm=a2g0s.9042311.0.0.27424c4dzvAtkF) and I noticed that the relay is **active low**. Here is the schematic: https://github.com/IOT-MCU/ESP-01S-Relay-v4.0/blob/master/ESP-01S%20Relay%20v4.0.pdf

Espurna's definition for this module (build flag GENERIC_ESP01S_RELAY_V40) is **active high**.

Has anyone else had this problem?

I previously bought another ESP-01S relay module (older version) and it is active high and works fine with Espurna compiled firmware (espurna-1.13.3-generic-esp01s-relay-40.bin).

Thank you,

Filipe Múrias

xoseperez, 2019-01-20:
Do you mean the same version has two different configurations?

virgil-av, 2019-02-17:
Have the same issue as @filipemurias this relay module is inverted for some reason so when the web admin is off it is actually on and vice versa.

Would be nice to have a feature where you can check a box to invert GPIO0, basically having a normal and inverted option in configuration window

closed 2019-02-21 by xoseperez