xoseperez/espurna #1506: Wish: Support WebThing API from mozilla

rzr, 2019-01-24:
Wish: Support WebThing API from mozilla

Then device could be connected (and controlled by) moziot gateway,

if feature is desired then it should be easy to add

Some hints at:

https://github.com/rzr/webthing-iotjs/wiki/MCU
https://discourse.mozilla.org/t/itead-sonoff-might-be-good-to-consider/25402

mcspr, 2019-01-26:
Some notes for posterity:

https://iot.mozilla.org/wot/ the spec itself. honorable mention of websockets interface

https://github.com/mozilla-iot/webthing-arduino/
https://github.com/mozilla-iot/webthing-arduino/blob/a8a2a9909701046cc1191217eee35ed67848a00f/ESPWebThingAdapter.h is a reference server, using ESPAsyncWebServer + mDNS.
I think it will conflict with WEB_SUPPORT=1 though, async server instance will be created twice and '/' is [already handled](https://github.com/xoseperez/espurna/blob/dev/code/espurna/web.ino#L407)

https://iot.mozilla.org/gateway/
https://github.com/mozilla-iot/gateway
https://github.com/mozilla-iot/thing-url-adapter does mdns discovery for the Gateway

Have not yet tried setting up the Gateway itself

rzr, 2019-01-27:
well you can differ with content accept to json vs html (default)

xoseperez, 2019-01-27:
Well, this is actually very interesting. As far as I understand I'd set up a webthings gateway locally (use the RPi image or even better: the docker project in https://github.com/mozilla-iot/gateway-docker) and connect to it via the Arduino API. Definitely worth implementing it.

madb1lly, 2019-11-15:
Hi @xoseperez,
Did you have any success with this?
Cheers :)

rzr, 2019-11-15:
If you didnt start this is something I can consider to contribute in upcoming weeks

madb1lly, 2019-11-25:
Hi @rzr, if you're asking me then no I didn't start anything. Obviously I can't speak for @xoseperez :)

rzr, 2019-12-12:
@xoseperez , can I self assign this issue if nobody started to implement this yet ?

andreabenini, 2019-12-16:
I've just started using webthing gw too and I've some appliances flashed with Espurna.
I'm just looking around to see what it takes to develop modules/addons/plugins for webthing because I'd like to adopt it as well, is there something I can do to help you ? 
I don't want to waste someone else's time or duplicate/fork the effort for the same module, what I'm basically trying to achieve is a minimal firmware capable of talking directly to webthing and keep the setup/configuration part on webthing.

rzr, 2019-12-16:
well so far I only used webthing gw along espruna devices using a custom MQTT bridge:

https://github.com/rzr/webthing-iotjs/wiki/MQTT

But I think a native REST API would be more flexible, so I created this issue. 

If anyone tried want to try to use webthing-arduino library let me know we can finish/do this together...

I will publish a patch soon or later

andreabenini, 2019-12-16:
Same like you, I was hoping to get some sort of direct support with a REST API instead of using mqtt. I'm currently dealing with python doc for webthing and I don't have experience with the webthing-arduino lib. I'm familiar with embedded devices and programming (mostly on AVRs + C, not that much with arduino toolchain) but I can help in some way if you need.