xoseperez/espurna #1513: New Device: King Art - King Q4 Cover switch

seea2005, 2019-01-29:
I would like to control by MQTT two cover switch which are shown on ewelink as king Q4:

[https://es.aliexpress.com/item/EWeLink-App-UE-Reino-Unido-Wifi-Smart-Touch-cortina-interruptor-de-Control-de-voz-por-Alexa/32956776611.html?spm=a2g0s.9042311.0.0.3fc163c0VL8sRB](https://es.aliexpress.com/item/EWeLink-App-UE-Reino-Unido-Wifi-Smart-Touch-cortina-interruptor-de-Control-de-voz-por-Alexa/32956776611.html?spm=a2g0s.9042311.0.0.3fc163c0VL8sRB)

From ewelink we can:
- Open (state of device --> ON)
- Close (state of device --> OFF)
- Stop (state of device --> Pause)
- Set a position (attribute setclose --> between 0 and 100)

Don't know really what else I have to send to develop it.

Thanks!

oscarsan1, 2019-01-29:
It seems compatible, it has IC ESP8285. You have to change the firmware for Spurna and map input number pins for buttons and output number pins for relays. It's no easy if you don't know how. Folow the instructions in the wiki.

seea2005, 2019-01-29:
I trust you it's no easy, too much for me right know as I can't either find how in the wiki... 🤣 I would need some more explanations...

ColinShorts, 2019-01-29:
You might need to consider the role of the nuvoton chip, I believe it's got a small amount of flash and may also be used for gpio expansion.

If it can be made to work, it'll be a much better match than Sonoff T1's that people are trying to use to control blinds/shutters.

oscarsan1, 2019-01-29:
Yes, it's looks better that Sonoff T1's for blinds at first look. Anyway I'm controling blinds with Sonoff T1 without a problem for now.

mclaudiopt, 2019-03-13:
h ithere...any solution for this?

Marcos1978, 2019-05-05:
Hello everyone, it is possible to send the official firmware of any of these blind control devices. Whatever the official Sonoff or other brand, a greeting.

jrzezak, 2019-05-07:
> Yes, it's looks better that Sonoff T1's for blinds at first look. Anyway I'm controling blinds with Sonoff T1 without a problem for now.

Hi, could you tell me what you do for controlling a curtain with sonoff T1. Your wiring and configuration.
How can you cope with the problem of pushing the up and down buttons simultaneously. Is there any way to manage only one at the time, especially in the case you accidentally touch both buttons at the same time.

namezero, 2019-05-25:
Any news on this ?
How to identify gpio ?

namezero, 2019-05-27:
I found the buttons PINS, but cant find the GPIO0 to flash it, any help to identify it ?

voodooo82, 2019-08-20:
Really? I have flash with tasmota but I don't know the pin x using. Tomorrow I send you the information for flashing

voodooo82, 2019-08-20:
For flashing collegate sw a gdn before connect the power... Use this guides https://github.com/arendst/Sonoff-Tasmota/wiki/KingArt

mcspr, 2019-08-20:
Referencing that wiki page, it looks like the exact [ITEAD_SONOFF_T1_3CH clone (see ..._PIN values)](https://github.com/xoseperez/espurna/blob/43a44cf96447dc2e04f0ef936573950df359889e/code/espurna/config/hardware.h#L828)
With some small differences regarding the touch panel controller and lack of 433RF

voodooo82, 2019-08-21:
The t1 3ch is very difference, I have 10 king art  n3(the t1 3ch CLONE) and I use tasmota for t1 3 ch and is perfect but for this is very different

DoraDeu, 2019-08-22:
> The t1 3ch is very difference, I have 10 king art n3(the t1 3ch CLONE) and I use tasmota for t1 3 ch and is perfect but for this is very different

Could you explain your statement? Do you use the Sonoff T1 3 ch konfiguration in Tasmota for your king art n3 devices?

voodooo82, 2019-08-22:
Yes I've flash tasmota with sonota and used with sonoff t1 3ch all my  king art n3(i have 10
[image]
[image]

 device)

voodooo82, 2019-08-22:
If you don't belive 😉 [image]

DoraDeu, 2019-08-22:
OK. So it works out of the box with the Sonoff T1 3CH setup in Tasmota.
Unfortunately I don´t have the 3 channel switch but the KingArt Curtain Switch from the beginning of this thread.
I managed to flash Tasmota on it but as the middle switch is not connected to a relay but is just a button to stop the up or down movement of the curtain I would need some help how to configure it in Tasmota and how to use it.

voodooo82, 2019-08-23:
I ve the same problem but more people using the dimmer panel over tasmota for control this device

mcspr, 2019-08-23:
That is great and all, but how can we fix Espurna firmware for this device? This is not Tasmota discussion forum 😕 
What exactly is missing for the curtain switch handling? Does this mean it has 3 buttons, but only 2 controllable switches instead of 3 (like the t1-3ch)?

voodooo82, 2019-08-23:
Espurna, tasmota is a firmware, the problem is the map of gpio if I know the gpio I can use all firmware

DoraDeu, 2019-08-23:
> That is great and all, but how can we fix Espurna firmware for this device? This is not Tasmota discussion forum 😕
> What exactly is missing for the curtain switch handling? Does this mean it has 3 buttons, but only 2 controllable switches instead of 3 (like the t1-3ch)?

You are right - sorry. I would also use Espurna if it would support the device.
The tricky thing is that this is not a normal 3 channel touch switch like Sonoff T1 3ch but it is a curtain switch. That means it only has two relays - one for closing and one for opening the curtain. The touch button in the middle is just a button to stop the motion. There is no relay needed. 
When I apply the Sonoff T1 3ch setup in Tasmota I can just control the LED in the corner. If all buttons are "off" the LED is on. If I turn one of the three buttons in Tasmota to on, the LED turns off. The LEDs on the buttons of the device do not change the color at all but this software switching. 
When I touch the "open" touch button on the device the LED turns to red for some seconds. When I press the same button again it changes to red for just a second. So the device knows that the curtain is open and does not try to open it again. The same thing happens with the "close" button. The "stop" button always changes to red for a second. So even after flashing with Tasmota there seems to be a program in place. Must have something to do with the other ICs on the board.

mcspr, 2019-08-23:
Are you using mains to power the device while testing?

Looks like that single LED is directly controlled by the ESP. Comparing with Digoo NX sp202 that I have with 6.5.0 (basic) version, LED depends on relay status from the SW side, ON when relay is OFF and OFF when relay is ON

There was another discussion about a similar device, where direct GPIO access seemed to work just fine https://github.com/xoseperez/espurna/issues/1855#issuecomment-522320136
Perhaps, it is finally a Tuya serial-controlled device, but i'd try that pin layout first

DoraDeu, 2019-08-23:
No, up to now I just powered it via the FTDI adapter but this is the next thing I am going to do.
Then I will check if the relays are controlled or not.
What is the normal behavior here if you control the switch via software: Does the switch status change the LEDs of the (3) touch switches or is this only the case if you use the touch switches manually?
Thank you for the link to a similar device! 
When you look here https://github.com/arendst/Sonoff-Tasmota/wiki/KingArt you find a very similar layout of a 1 ch switch with an IC named TW8001 from TechWave but I have no idea what it does on the board.
The Tuya devices normally run with SmartLife or Tuya cloud but the KingArt has ewelink firmware.

DoraDeu, 2019-08-25:
I checked it with mains power and it works the same - you can completely control the switch with the touch buttons and the relays are switching but with the Tasmota web interface just the control LED at the corner of the board reacts if the switches are used.

mcspr, 2019-08-27:
LED is working based on software state and it happens at the same time it tries to write relay (either 0 or 1, based on wanted state). Running Espurna you'd get the same LED behaviour, as we don't get any feedback from relay and only assume that it is on. Touching buttons should trigger button action though, despite that relay pin is likely wrong. Maybe something in the logs?
What you can try right now is to use "Generic" module while running Tasmota to setup pins from the other issue

voodooo82, 2019-08-27:
I have try with generic but I Don, t know the gpio...

voodooo82, 2019-08-29:
In the tasmota page a new device kingart dimmer is very similar and with this configuration the device work... But isn't perfect 
[image]

voodooo82, 2019-09-01:
Isn't perfect because I have only one button, this button in ever On if I switch in Off the curtain closed but if I want stop and repush in On the curtain reopen for 2 o 3 second. If I muve the cursor only open but I don't know for how many time... 
[image]

DoraDeu, 2019-09-02:
That ´s a tricky one! As the ESP communicates with the switching IC via serial commands we don´t get it working by just assigning GPIOs correctly.
For the dimmer someone found the proper communication:
[https://github.com/arendst/Sonoff-Tasmota/wiki/PS-16-DZ-Dimmer](url)
For the curtain switch the commands must be similar but I don´t know how to find them out. Maybe it is possible to sniff the communication on RX/TX while the buttons are pressed on the device.

corvidus, 2019-11-09:
Hi all, 
First post, not a pro-coder, pls be gentle blah blah blah. 
I've been stuck in the same loop as everyone else here, trying to get this particular device to play nice as a motor control/cover/blind/shutter/curtain/callitwhatyouwant switch. After trawling a bunch of different forums, I got annoyed enough to rip mine apart, took @DoraDeu suggestion above, attached a RS232 UART and did some poking around with a Serial Monitor. The commands ARE indeed similar. Sample output as follows: 

Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 6e 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 31 30 30 1b                                      
AT+UPDATE="switch":"on","setclose":100.
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 6e 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 36 35 1b                                           
AT+UPDATE="switch":"on","setclose":65. 
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63  68 22 3a 22 6f 6e 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 32 38 1b                                           
AT+UPDATE="switch":"on","setclose":28. 
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63   AT+UPDATE="switch":"on","setclose":0.
68 22 3a 22 6f 6e 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 30 1b                                                
Read data (COM4) 
 41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 6e 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 30 1b 
AT+UPDATE="switch":"on","setclose":0. 
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 66 66 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 30 1b 
AT+UPDATE="switch":"off","setclose":0.  
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 70 61 75 73 65 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 31 38 1b 
AT+UPDATE="switch":"pause","setclose":18.
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 66 66 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 31 38 1b  
AT+UPDATE="switch":"off","setclose":18. 
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 66 66 22 2c 22 73 65 74 63 6c 6f   
73 65 22 3a 35 34 1b 
AT+UPDATE="switch":"off","setclose":54. 
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 66 66 22 2c 22 73 65 74 63 6c 6f   
73 65 22 3a 39 31 1b 
AT+UPDATE="switch":"off","setclose":91. 
Read data (COM4) 
41 54 2b 55 50 44 41 54 45 3d 22 73 77 69 74 63 68 22 3a 22 6f 66 66 22 2c 22 73 65 74 63 6c 6f 73 65 22 3a 31 30 30 1b 
AT+UPDATE="switch":"off","setclose":100.         `

Looks fairly straight-forward really. "switch":"on" goes up/open, "switch":"off" goes down/closed, and "switch":"pause" seems to be the missing link for halting the motor without causing the switch to turn "off" and therefore down/closed again. 

Now I'm not particularly well versed with code, but I've knocked out a version of the PS-16-DZ xdrv file (attached). I've not compiled it (mostly because I expect it would need to be integrated with the rest of the code in ways I'm so not in the mood to work out right now), so consider it "untested". I'm also not sure whether it would be better integrated with the xdrv_27_shutter module. I'm hoping someone who knows more about these things than I do can take this and run with it, but hopefully I've done the worst of the legwork here. 

Separately, I've attached the Template string that I've been using for this - the only difference from the PS-16-DZ Template is that the LED on GPIO13 is inverted so that SetOption31 and LedPower commands work as intended: 

{"NAME":"KingArt Cover","GPIO":[255,148,255,149,255,255,0,0,255,56,255,255,255],"FLAG":0,"BASE":58}

Happy hacking! 

[xdrv_99_ka_cover.txt](https://github.com/xoseperez/espurna/files/3827031/xdrv_99_ka_cover.txt)

erelor, 2019-12-01:
I have the same device and it would be great if someone could integrate it in the next version of the official firmware

mcspr, 2019-12-01:
Note that the PS16DZ original driver mentioned above is here:
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_19_ps16dz_dimmer.ino
Changes are hard to follow though, because of renames :/ But that was Tasmota stuff discussed for whatever reason.

I guess we could add some new device driver, separate from lights / dimmer, that will have 0...100 range for possible values + a separate direction switch
Maybe could also be useful for iFan-like stuff, showing direct scale for the speed instead of relays emulating 25, 50, 75 and 100%

closed 2020-05-01 by mcspr

cavern-man, 2020-05-18:
> I found the buttons PINS, but cant find the GPIO0 to flash it, any help to identify it ?

Hello, I'm slowly digging into the ESPHome maze to use with King Q4. By the way, did you manage to work with king Q4 or do you recall the GPIO map ?

Many thanks in advance.

psasin, 2021-03-20:
> 
> 
> Really? I have flash with tasmota but I don't know the pin x using. Tomorrow I send you the information for flashing

How did You manage to flash KingArt Q4? I've tried several times and I can't flash it. Serial connection seems to be OK, because when I connect with Hercules I can see some commands coming from the device, but I'm not able to get it in the flashing mode. Shorting GND to pad designated as SW on PCB, but no success. I've even measure that pas SW is connected to GPIO0 on ESP8285.
I'm using Tasmotizer 1.2 + CP1204 and it is working fine for Sonoff devices but not for KingArt.

Here is photo of my setup:
[image]

mcspr, 2021-03-21:
@psasin you should be able to use serial monitor apps to see [ESP bootloader messages](https://arduino-esp8266.readthedocs.io/en/latest/boards.html#boot-messages-and-modes), which are usually sent out with 74480 baudrate. e.g. this should come up when chip is ready to flash when it is just had been powered on:
```
> pio.exe device monitor --baud 74480  --echo
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  74480,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```

And note that you should disconnect GPIO0 to be able to actually flash the board

daniel3514, 2021-11-10:
greetings to all! I have found GPIO0, I marked it in a circle, the track to be opened is marked in the square, because it is high and that does not allow flashing, even so, I still have problems, it does not connect 
[image]

daniel3514, 2021-11-10:
this could help:

https://github.com/arendst/Tasmota/issues/5059

4139GGN, 2022-12-20:
Hello!
As of today, does anyone have the King Art q4 running with Tasmota or with some other Firm?
I have already tried several and I have not been able to control the blinds from home assistant for several months...