xoseperez/espurna #1515: Platform.io. one task for each device?

oscarsan1, 2019-01-31:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

oscarsan1, 2019-01-31:
I do OTA download with platform.io folow the instructions in the wiki. Finaly my task.json is:

    "version": "2.0.0",
    "tasks": [
        {
            "type": "PlatformIO",
            "label": "Upload Arbol",
            "task": "Upload (itead-sonoff-basic-ota)",
            "options": {
                "env": {
                    "ESPURNA_IP": "192.168.1.200",
                    "ESPURNA_AUTH": "-----"
                },
            },
            "problemMatcher": [
                "$platformio"
            ]
        }
    ]
can I define more tasks, one for each device sonoff basic I have, with diferents  IP each device but the same task "Upload (itead-sonoff-basic-ota)"?

mcspr, 2019-02-04:
*note: not a heavy vscode user*

I did try adding multiple tasks referencing `task: upload (...)` but it just lists the last one 🤷‍♂️ 
Possible solution: use `type: shell` and manually type out required `platformio command`, check out [tasks.json docs](https://code.visualstudio.com/docs/editor/tasks)
e.g. new task should have `"type": "shell", "command": "platformio run -e itead-sonoff-basic-ota -t upload"`

oscarsan1, 2019-02-05:
That's works.  Thank you. 
I do in task.jsom:

        {
            "label": "PlatformIO: Arbol1",
            "type": "shell",
            "command": "platformio run -e itead-sonoff-basic-ota -t upload",
            "options": {
                "env": {
                    "ESPURNA_IP": "192.168.1.200",
                    "ESPURNA_AUTH": "---"
                },
            },
            "problemMatcher": [
                "$platformio"
            ],
        },
        {
            "label": "PlatformIO: Light1",
            "type": "shell",
            "command": "platformio run -e itead-sonoff-basic-ota -t upload",
            "options": {
                "env": {
                    "ESPURNA_IP": "192.168.1.201",
                    "ESPURNA_AUTH": "---"
                },
            },
            "problemMatcher": [
                "$platformio"
            ],
        }

With this I can execute each task "Arbol1" and "Ligth1" for OTA download code to each device.

closed 2019-02-05 by oscarsan1