xoseperez/espurna #1516: Domoticz and Hass tabs not shown

julgonmej, 2019-01-31:
I built Espurna from source some months ago (I don't know the commit but it's shown as ESPURNA 1.13.1a). I used the default options, I just added some switches and relays. Domoticz, Hass, Schedule and NTP tabs were shown in the web.
Now, I need to modify the firmware to add another relay, but I noticed that those tabs are not shown anymore. I tried enabling all of them (HOMEASSISTANT_ENABLED 1 and so on) but to no avail.
Have those pages been deleted?

julgonmej, 2019-02-04:
This issue is not new, I tried back in November with the same results.
Right now I was reproducing it with the current dev branch. `1874f28`.  I tried with the solution provided on #1457 (`#define WS_MAX_QUEUED_MESSAGES 16`) and it seems to work

mcspr, 2019-02-04:
Please do try with default queue length and changes from #1387 
(see https://help.github.com/articles/checking-out-pull-requests-locally/) 

Note that webui needs to be rebuilt using `gulp`

closed 2019-02-09 by xoseperez