xoseperez/espurna #1522: Shield DHT22 con ESp8266 - ESP01

pablopescador, 2019-02-04:
Hola que tal, 
Quisiera saber si es que puedo usar su Firmware para el siguiente Shield
[image]

Estoy sufriendo con el platformio.ini 
para declarar algo asi por ejemplo? env_default = generic-esp01s-dht22-22
y el en sensor.h

#ifndef DHT_SUPPORT
#define DHT_SUPPORT                     1
#endif

#ifndef DHT_PIN
#define DHT_PIN                         2
#endif

#ifndef DHT_TYPE
#define DHT_TYPE                        DHT_CHIP_DHT22
#endif

lo he puesto asi. 
pero cuando "build" me sale error.
Si uso el el nombre aceptado por defecto en platformio.ini
como env_default = generic-esp01s-dht11-11
va todo bien, pero no lee el sensor y me sale esto
[image]

Muchas gracias por compartir @xoseperez.
Un saludo desde Alemania.

xoseperez, 2019-02-07:
Es un DHT22, no un DHT11, la imagen que has utilizado es para el DHT11.
Deberás compilarlo tal como has definido arriba, qué error te da?

pablopescador, 2019-02-11:
Hola Xose, muchas gracias por tu tiempo. Ya lo podido compilar. Va de lujo tu firmware. Ahora estoy peleando con mqtt broker en un 8266 a ver que sale ;) ...

closed 2019-02-11 by pablopescador

gregoriohc, 2019-12-04:
Hola @pablopescador. ¿Tienes de casualidad por ahí el bin para el DHT22? Me ahorrarías el trabajo de tener que instalar todo lo necesario para compilarlo yo y te estaría eternamente agradecido, jeje.
Mil gracias y también a @xoseperez por el gran trabajo en su firmware!

pablopescador, 2019-12-04:
Hola Gregorio,
Lo siento pero lo logre hacer funcionar y lo borre :(
Saludos.

gregoriohc, 2019-12-05:
Gracias @pablopescador. Me pondré a ello entonces :-)