xoseperez/espurna #1523: device description as part of the device configuration

wsw70, 2019-02-04:
**Is your feature request related to a problem? Please describe.**

A device holds a hostname, which I use in a generic way (`switch-1`, `switch-2`, ...). This way I can go though my switches (and other devices named in a similar way) and update, reboot, inventory them.

A device has however also a practical denomination (`small lamp in room 1`) which does not have a natural place. Today I use the MQTT topic for that (`rm1-lamp-small`).

**Describe the solution you'd like**

It would be interesting to have a field, in the surroundings of  `hostname`, to hold the description. A string would be great.

**Describe alternatives you've considered**

MQTT topic per above.

This is not the most important feature in the world but it could be quite helpful still.

mcspr, 2019-02-06:
I would even put this issue with #1429, `SHP2/4/6` name can be default for this kind of "friendly name" setting.
Name itself should go to MQTT (most likely `<root>/name`), mDNS announce, it would be somewhere on the Web starting page. Preferably as Alexa name too, although I think there was a different proposal having separate custom names for each switch instead (`Left lamp`, `Right lamp` instead of `Switch #1` and `Switch #2`).

xoseperez, 2019-02-07:
Not sure this is the same. What will we do with this "description" field beyond having it. Maybe sending it with the MQTT heartbeat?

wsw70, 2019-02-07:
> What will we do with this "description" field beyond having it

The problem I was trying to solve with this is that I have a generic name for the devices in order to name them in an uniform way and be able to always refer to them the same way (for instance `switch-3` has a big `3` written on its back and I can always look at it and know that this is `switch-3`).

This generic name is very helpful to iterate over devices in order to do something (update them, mostly - but also monitor for instance).

This generic name is not very practical, though, for "interactive" uses. I prefer to know that `room1-lamp-ontheside` has sent an MQTT message, rather than `switch-3`. Sure, I could find out but when my wife tells me that the lamp in the living room does not work AGAIN and I am in Barcelona and not in Versailles, I prefer to be able to quickly connect to `http://room1-lamp-ontheside.at.home`  and switch it on.

This is why I am creating `CNAME` entries for each generic entry and, ideally, would be glad to automatize this (by querying the description and deriving a `CNAME` from it, and attach it to the hostname).

The other usage is that even without the `CNAME`, I would be happy to be able, by just connecting to `switch-27` to be able to say that this is the switch in the bedroom on the right. Or having my monitoring system warn me not that `switch-7` is down but that `bedroom parents lamp left` is down (taken from the description), which means crisis, vs `bedroom children lamp ontheside`, which means never mind they do not use it anyway.

Like I mentioned in my initial question, this is a nice to have, not a core functionality. Today I deal with by using the MQTT topic as the "meaningful name" (and hostname as the "generic name") and this is fine. I was just wondering if anybody else had the same need (apparently not). 

I will be glad to close this if there is no interest, this was really a low priority nice to have. The fact that everything else works great is by far the most important (and thanks again for that super firmware)

xoseperez, 2019-02-07:
I think it's a good idea to have a description for each device, something more human friendly. Sure it's not a must have but don't close it.

> by querying the description and deriving a CNAME from it,

How would you (ideally) query the description?

wsw70, 2019-02-07:
> How would you (ideally) query the description?

Either via tha REST API or MQTT (via the heartbeat message you mentioned).

Since I am actively moving towards Home Assistant with all my automations (and probably replacing the ugly family dashboard I developed myself...), MQTT may be more relevant as it is easily manageable (= it is easy to add a "sensor" which would take the friendly name and build something from it)

closed 2019-02-10 by xoseperez

wsw70, 2019-02-10:
Thank you very much!