xoseperez/espurna #1524: RGB lights sending wrong Format value?

vMeph, 2019-02-04:
im using Airlux AL_LC01 RGB leds with Espurna 1.13.0 with homeassintant
in homeassitant it works perfect, but in dev i always have seen this  error
```
Exception in rgb_received when handling msg on 'RGB/rgb': '#FFC824'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/mqtt/light/schema_basic.py", line 315, in rgb_received
    rgb = [int(val) for val in payload.split(',')]
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/mqtt/light/schema_basic.py", line 315, in <listcomp>
    rgb = [int(val) for val in payload.split(',')]
ValueError: invalid literal for int() with base 10: '#FFC824'
```
so i question balloob about  this erros and acording to it, the problem has to do with the way its sending values, in this case we are sending HEX values and it should be formated like 255,200,120

is there anything that could be done to prevent

ColinShorts, 2019-02-05:
Do you have CSS colours switched on by any chance?

vMeph, 2019-02-05:
@ColinShorts hehe you the man, wow that fix the problem, i could i have miss that :)
thank you !!!!

closed 2019-02-05 by vMeph

mcspr, 2019-02-06:
Also see #1273 
Either HA should parse `#HEXHEX` values too or `Use CSS` option should send data to `<root>/css` topic exclusively and leave rgb to the comma separated values

perikillo70, 2020-02-12:
Hello, I has same problem. It works but in logs appear :

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 362, in rgb_received
    rgb = [int(val) for val in payload.split(",")]
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 362, in 
    rgb = [int(val) for val in payload.split(",")]
ValueError: invalid literal for int() with base 10: 'FF000000'

2020-02-12 10:04:58 ERROR (MainThread) [homeassistant.util.logging] Exception in rgb_received when handling msg on 'tele/bulb1/STATE': '{"Time":"2020-02-12T10:04:58","Uptime":"4T14:19:51","UptimeSec":397191,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":100,"Color":"FF000000","HSBColor":"0,100,100","White":0,"Channel":[100,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"MOVISTAR_12E8","BSSId":"E0:41:36:77:54:BB","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:06"}}'