xoseperez/espurna #1527: ]Sonoff DUAL R2

scargill, 2019-02-06:
IU'm completely new to Espurna but not MQTT, Sonoff etc.

I just installed latest release espurna for Dual R2. Wifi is up and running - WIFI interface I can turn relays on and off. 

MQTT - given the root topic "simples" - I've tried sending a simple message.. topic..
simples/relay/1/set  and numerical payload of 0 and 1 - nothing happens.

Also, in WIFI setup, switches interface I set both switches to same as before and not syncronised. But on power off then back on, the relay states are not holding over duing power down.

Most or all of this could be me - new as in 2 hours ago... any pointers would be great.  I use Node-Red and MQTT so no hass, domotics etc, I like to keep it simple.

Yes I've set up the mqtt ip, user name and pass also root topic simples.

Do I have to do more?

fragolinux, 2019-02-06:
same issues here, sonoff dual r2, same firmware, flashed using https://github.com/BattloXX/ESPEasyFlasher

they were both on, i put them off, it didn't ask to save, on reboot i got switch#1 on 
i put both on, pressed reboot, it ASKED if i wanted to save this time, i pressed cancel and rebooted, #0 off and #1 on...
inverted them so #01 was on and #1 off, rebooted without saving, both off...
weird... no sync between them enabled, and set them as "as before"

```
---8<-------

[603677] [MAIN] ESPURNA 1.13.3 (e2641b9)
[603678] [MAIN] xose.perez@gmail.com
[603679] [MAIN] http://tinkerman.cat

[603680] [MAIN] CPU chip ID: 0x9D6FB7
[603682] [MAIN] CPU frequency: 80 MHz
[603686] [MAIN] SDK version: 1.5.3(aec24ac9)
[603690] [MAIN] Core version: 2.3.0
[603693] [MAIN] Core revision: 159542381
[603696] 
[603697] [MAIN] Flash chip ID: 0x144051
[603701] [MAIN] Flash speed: 40000000 Hz
[603704] [MAIN] Flash mode: DOUT
[603707] 
[603708] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[603716] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[603725] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[603730] [MAIN] Firmware size       :   463808 bytes /  114 sectors (   1 to  114)
[603737] [MAIN] Max OTA size        :   557056 bytes /  136 sectors ( 115 to  250)
[603744] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[603751] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[603759] 
[603761] [MAIN] EEPROM sectors: 251, 250
[603763] [MAIN] EEPROM current: 250
[603766] 
[603767] [MAIN] EEPROM:  4096 bytes initially |  1133 bytes used (27%) |  2963 bytes free (72%)
[603778] [MAIN] Heap  : 36848 bytes initially | 19424 bytes used (52%) | 17424 bytes free (47%)
[603784] [MAIN] Stack :  4096 bytes initially |  1272 bytes used (31%) |  2824 bytes free (68%)
[603792] 
[603793] [MAIN] Boot version: 6
[603795] [MAIN] Boot mode: 1
[603798] [MAIN] Last reset reason: Reboot from web interface
[603803] 
[603804] [MAIN] Board: ITEAD_SONOFF_DUAL_R2
[603808] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[603825] [MAIN] WebUI image: SMALL
[603826] 
[603827] [MAIN] Firmware MD5: 1695693c3d37e7bf8c5e331f3b10d4a3
[603833] [MAIN] Power: 3183 mV
[603835] [MAIN] Power saving delay value: 10 ms
[603839] 

---8<-------
```

scargill, 2019-02-06:
I just went back to 1.32 - now works on reboot - but strangely still offers to save (web interface - status - switch change state) on reboot. If I DON'T save, it still saves status as it should.

fragolinux, 2019-02-06:
same here, going back to 1.13.2 seems to have solved the issue, but it asks to save if i just use the web gui to click the switches, but not always! if i put both off it does not ask to save, any other combination, yes... i press REBOOT, and never confirm the saving, pressing cancel when it asks...

scargill, 2019-02-06:
Also when I moved back to 1.13.2 like you, my "test lamp" Alexa moved from "test lamp 1" and "test lamp 2" to "test lamp 0" and "test lamp 1" leaving "test lamp 2" as an orphan.

mcspr, 2019-02-06:
The problem is relays are "on" regardless of any setting?
If so, pls see https://github.com/xoseperez/debounceevent/pull/10 and #1254

scargill, 2019-02-06:
1.13.3 fails badly (as above) so we went to 1.13.2 - that works - but when changing button state via WIFI status page - you are sometimes asked if you want to SAVE this. That isincorrect as it is already saved on reboot (SWITCHES setting SAME AS BEFORE) so there is nothing to save. Other issue is change of relay numbering between 1.13.2 (0 and 1) and 1.13.3 (1 and 2).

xoseperez, 2019-02-07:
I have updated the DebounceEvent library to 2.0.5 and also the reference in the PlatformIO file. If you are using Arduino IDE you should update it from the Library Manager (hopefully it will be available soon). This should account for the wrong on boot states.

About the relay numbering change this is happening with Alexa, right? Sorry about that, the fauxmoESP library it uses changed from 2.X to 3.X and it looks like it had this effect...

I'm checking why the UI asks to save even if there has been no change...

fragolinux, 2019-02-07:
thanks, and will you release new binaries, too?

xoseperez, 2019-02-07:
Soon

xoseperez, 2019-02-21:
Changes are being released with 1.13.4, but I'm not closing this until I get an OK from you.

mcspr, 2019-10-22:
Closing via https://github.com/xoseperez/debounceevent/pull/10
If web save is still a problem on current dev version (not 1.13.2! as noted above), please open a new issue

closed 2019-10-22 by mcspr