xoseperez/espurna #1528: Error compiling Espurna in Arduino

AshokBaijal, 2019-02-06:
While compiling Espurna in Arduino IDE with NodeMCU I get following error

"In file included from sketch\config/all.h:29:0,

                 from C:\Users\baija\Desktop\espurna-dev\code\espurna\espurna.ino:22:

sketch\config/hardware.h:3498:6: error: #error "UNSUPPORTED HARDWARE!!"

     #error "UNSUPPORTED HARDWARE!!"

      ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module)."

What is the solution? Please help.

xoseperez, 2019-02-07:
Check the wiki (https://github.com/xoseperez/espurna/wiki/ArduinoIDE#configuring-the-hardware), you have to enable at least one target board before building it.

closed 2019-02-07 by xoseperez

AshokBaijal, 2019-02-07:
Thanks. Was able to compile as I realised my mistake of not uncommenting the #define NODEMCU_LOLIN directive but got stuck with errors relating to NTPClientLib. Got solved after renaming the Library folder from NTPClent-develop to NTPClientLib.h