xoseperez/espurna #1530: Default relayBoot/Pulse/TimeN settings with many switches

mcspr, 2019-02-06:
https://github.com/xoseperez/espurna/blob/1874f28dc4922d77162d5fba90a181921856c45b/code/espurna/relay.ino#L479-L498

Boards like RFBridge with 20+ switches result in EEPROM corruption, see https://github.com/xoseperez/espurna/issues/262#issuecomment-457970954

xoseperez, 2019-02-07:
This was added more than one year ago (https://github.com/xoseperez/espurna/commit/b3336871bf70e57962f16c03a2d67a69e15fd14d) so I guess it is mostly save to delete, but:

1) why is it recreating the settings on every loop? the `hasSetting` check should return false and do nothing...
2) it could be enough to test if `hasSetting("relayMode")` and exit the method if false in the very begining.

mcspr, 2019-02-07:
I meant that they persistently come back when deleted, because loop is executed every time. Checking if old key exists does sound like a solution. BTW, can `cfg` be used more extensively for such cases? i.e. bump it every time any config key is changed and wire ...Backwards-like methods right after settings storage is loaded.

Another thing - web will send them back and fill the settings again! On the same multi-switch board, `getData($(".form-settings"))` will gather the same keys as relay...N=0. Now I remember f59d747af1b43dc3903305ab69f76b6d1c3594d3 trying to add 'original' protection, but it added another bug so all most of the inputs have empty 'original' 😨 
- OnStart methods can be marked somehow and reset attrs only in that case, repeating the original intent and avoiding resets on weblog / status updates / etc.
- since 'original' value now can be trusted, getData can filter out based on 'hasChanged' attribute and only send small diff instead of whole config

xoseperez, 2019-02-09:
Yes, we surely could be updating `cfg` more often. It will also require centralizing all migrations in a single module (`migrate.ino`?).

mcspr, 2019-02-09:
Nah, I think just passing the `cfg` value should be enough. So the flow can be
setup() calls `settingsSetup()`, gets the cfg value, records it
every other setup() calls `backwards(settingsConfigVersion())`, migrates based on that
setup() calls `settingsConfigVersion(CFG_VERSION)` (or some hypothetical scheduler calls it implicitly just before first ever loop(), only once)

Some interesting things can be done using global object constructors - make some kind of Settings class, subclass in every other module. Now original constructor can record subclassed module object pointer into shareable static array, so settingsSetup() can then call them directly.

closed 2019-11-09 by mcspr