xoseperez/espurna #1532: BLITZWOLF BWSHP2 V23 & V24

AlbertWeterings, 2019-02-07:
BLITZWOLF_BWSHP2_V23 is already supported in hardware.h but a line to enable it in arduino.h is missing.
BLITZWOLF_BWSHP2_V23 setup is also working for V24 board.

//#define BLITZWOLF_BWSHP2_V23

xoseperez, 2019-02-07:
But maybe it should be changed to BLITZWOLF_BWSHP2_V2X

AlbertWeterings, 2019-02-07:
Might be but it is not sure it will also work for 2.1, 2.2 and possible revisions 2.5 and higher so not sure this good to do.

xoseperez, 2019-02-07:
True, anyway this naming convention is a problem.

AlbertWeterings, 2019-02-07:
I agree, also tried to figure out a different way as  I think in the future we are going to see more and more hardware with different revisions. So maybe breaking up the naming like ((general devicename)(revision)) is an option and then a structure with sub parts in hardware.h is an option.

closed 2019-02-10 by xoseperez