xoseperez/espurna #1537: Adding Armtronix NodeMCU_four_relay_board module: Nothing happens

aglproyectos, 2019-02-08:
I am the fortunate owner of a Armtronix NodeMCU_four_relay_board with which I have already tested in Arduino context and developed an application involving web service and MQTT in real operation.
Now I am trying to implement Espurna that seems very promising as it includes may other features which most of them I am interested like NTP and scheduling but, first things first, I juast have started with web support and MQTT support.

As a starting point I have used NodeMcu BASIC  definition including on it definition of the four relays and one led on the board (as they are defined in SONOFF 4CH) just changing the GPIO pin numbers to the appropriate values Arduino.h and Hardware.h

The problems I get are the following ones:
- The WiFi access sometimes appears sometimes not and I can not relate that fact with a specific key or value change.
- Sometimes finally I get the web page with the board status but it never shows the relays nor the led.
- The web page does not show expected MQTT tab or any other tha WIFI, STATUS, ... nothing related to other Espurna features.

Is it too easy from my side just to modify Arduino.h and Hardware.h?
What else do I have to do?

xoseperez, 2019-02-09:
Hi, answering your issues one by one:

- You should check if the board is booting correctly, use the serial debug to check if there is any problem. 
- Can you copy the configuration you are using? 
- There is an issue with the latest version of ESPAsyncWebServer (see #1457). We have requested a change to the developer and we are testing a work around at the same time. But in the meantime try the change suggested in #1457 or use an older version of the library (before August 2018).

aglproyectos, 2019-02-09:
Hi, thanks for that quick answer...

The serial monitor shows the board is rebooting every few (around 8) seconds as follows:

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000088] 

---8<-------

[000088] [MAIN] ESPURNA 1.13.4-dev
[000088] [MAIN] xose.perez@gmail.com
[000088] [MAIN] http://tinkerman.cat

[000089] [MAIN] CPU chip ID: 0xC78D71
[000092] [MAIN] CPU frequency: 80 MHz
[000095] [MAIN] SDK version: 1.5.3(aec24ac9)
[000099] [MAIN] Core version: 2.3.0
[000102] [MAIN] Core revision: 159542381
[000106] 
[000107] [MAIN] Flash chip ID: 0x1640EF
[000110] [MAIN] Flash speed: 40000000 Hz
[000114] [MAIN] Flash mode: QIO
[000116] 
[000118] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[000125] [MAIN] Flash size (SDK)    :  4194304 bytes / 1024 sectors (   0 to 1023)
[000132] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[000139] [MAIN] Firmware size       :   469824 bytes /  115 sectors (   1 to  115)
[000146] [MAIN] Max OTA size        :  2670592 bytes /  652 sectors ( 116 to  767)
[000153] [MAIN] SPIFFS size         :  1028096 bytes /  251 sectors ( 768 to 1018)
[000161] [MAIN] EEPROM size         :     4096 bytes /    1 sectors (1019 to 1019)
[000168] [MAIN] Reserved            :    16384 bytes /    4 sectors (1020 to 1023)
[000175] 
[000176] [MAIN] EEPROM sectors: 1019, 1018, 1017, 1016
[000181] [MAIN] EEPROM current: 1017
[000184] 
[000185] [MAIN] EEPROM:  4096 bytes initially |   292 bytes used ( 7%) |  3804 bytes free (92%)
[000193] [MAIN] Heap  : 36880 bytes initially |  5304 bytes used (14%) | 31576 bytes free (85%)
[000202] [MAIN] Stack :  4096 bytes initially |   928 bytes used (22%) |  3168 bytes free (77%)
[000210] 
[000210] [MAIN] Boot version: 31
[000213] [MAIN] Boot mode: 1
[000216] [MAIN] Last reset reason: Power on
[000220] [MAIN] Last reset info: flag: 0
[000223] 
[000224] [MAIN] Board: NODEMCU_BASIC
[000227] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[000242] [MAIN] WebUI image: SMALL
[000245] 
[000428] [MAIN] Firmware MD5: 2545a847d77224073e144742d9751796
[000429] [MAIN] Power: 2534 mV
[000429] [MAIN] Power saving delay value: 10 ms
[000429] 

---8<-------

[000437] [TELNET] Listening on port 23
[000438] [WEBSERVER] Webserver running on port 80
[000442] [RELAY] Retrieving mask: 0
[000443] [RELAY] Relay #0 boot mode 0
[000446] [RELAY] Relay #1 boot mode 0
[000449] [RELAY] Relay #2 boot mode 0
[000452] [RELAY] Relay #3 boot mode 0
[000456] [RELAY] #0 set to OFF
[000458] [RELAY] #1 set to OFF
[000461] [RELAY] #2 set to OFF
[000464] [RELAY] #3 set to OFF
[000467] [RELAY] Number of relays: 4

**The configuration I have defined in Hardware.h is:**
#elif defined(NODEMCU_BASIC)
    // Info
    // Generic NodeMCU Board without any buttons or relays connected.
    #define MANUFACTURER        "NODEMCU"
    #define DEVICE              "BASIC"
    // Buttons
    // No buttons on the D1 MINI alone, but defining it without adding a button doen't create problems
//    #define BUTTON1_PIN         0   // Connect a pushbutton between D3 and GND,
                                    // it's the same as using a Wemos one button shield
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1
    #define BUTTON1_PIN         9
    #define BUTTON2_PIN         2
    #define BUTTON3_PIN         10
    #define BUTTON4_PIN         16

//    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON4_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON1_RELAY       1
    #define BUTTON2_RELAY       2
    #define BUTTON3_RELAY       3
    #define BUTTON4_RELAY       4

    // Relays
    #define RELAY1_PIN          4
    #define RELAY2_PIN          12
    #define RELAY3_PIN          13
    #define RELAY4_PIN          14

    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL
    #define RELAY3_TYPE         RELAY_TYPE_NORMAL
    #define RELAY4_TYPE         RELAY_TYPE_NORMAL

**And, arduino.h the following features:**

//--------------------------------------------------------------------------------
// Features (values below are non-default values)
//--------------------------------------------------------------------------------

//#define ALEXA_SUPPORT          0
//#define API_SUPPORT            1
#define BROKER_SUPPORT         1
#define BUTTON_SUPPORT         1
#define DEBUG_SERIAL_SUPPORT   1
//#define DEBUG_TELNET_SUPPORT   0
//#define DEBUG_UDP_SUPPORT      1
#define DEBUG_WEB_SUPPORT      1
//#define DOMOTICZ_SUPPORT       0
//#define ENCODER_SUPPORT        1
//#define HOMEASSISTANT_SUPPORT  0
//#define I2C_SUPPORT            1
//#define INFLUXDB_SUPPORT       1
//#define IR_SUPPORT             1
//#define LED_SUPPORT            1
//#define LLMNR_SUPPORT          1  // Only with Arduino Core 2.4.0
//#define MDNS_CLIENT_SUPPORT    1
//#define MDNS_SERVER_SUPPORT    0
#define MQTT_SUPPORT           1
//#define NETBIOS_SUPPORT        1  // Only with Arduino Core 2.4.0
//#define NOFUSS_SUPPORT         1
//#define NTP_SUPPORT            1
//#define RFM69_SUPPORT          1
//#define RF_SUPPORT             1
//#define SCHEDULER_SUPPORT      1
//#define SENSOR_SUPPORT         1
//#define SPIFFS_SUPPORT         1
//#define SSDP_SUPPORT           1
//#define TELNET_SUPPORT         0
//#define TERMINAL_SUPPORT       0
//#define THINGSPEAK_SUPPORT     0
//#define UART_MQTT_SUPPORT      1
#define WEB_SUPPORT            1

I'll have a look to the problem related to ESPAsyncWebServer meanwhile.

Once again, tanks in advance .

xoseperez, 2019-02-10:
Looks like the WDT reset might be related to the button configuration (they are initialized just after the relays). I see you are using GPIO16 which is not exactly general purpose...

closed 2019-04-18 by stale[bot]