xoseperez/espurna #1541: Global loop delay

mcspr, 2019-02-09:
https://github.com/xoseperez/espurna/blob/c4ad96b2fbd815798468904381b4df3b90ddee32/code/espurna/system.ino#L150-L154
- is there any reason not to do things as fast as possible? https://arduino.stackexchange.com/questions/49824/does-a-delay-in-a-loop-save-energy claiming otherwise. essentially, it only slows down the loop() code. sdk stuff is still running the wifi, async libs still handle requests.

xoseperez, 2019-02-10:
Back when it was introduced a user was testing the overall consumption and he found out there was a significant save introducing the delay... It all started with the same approach in Tasmota (https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/272)

oscarsan1, 2019-02-16:
Hi. I don't see any reducction in consumption, only a little increase with LOOP_DELAY_TIME =0, otherwise the consumption is the same 8,1 mA (0,8W). Each relay comsumption in on is 2 mA (aprox 0,3W). Cos phi 0,45 capacitive for translate mA to active power in W.
I guess that delay() is only a bucle of "nops", there is no sleep.
The response time is not good at all with 250 ms. I use then 1 ms in my devices.

mA | LOOP_DELAY_TIME
8,1 | 250
8,1 | 10
8,1 | 5
8,1 | 1
8,3 | 0

oscarsan1, 2019-02-16:
Is explained here: https://blog.creations.de/?p=149
delay() doesn't sleep the CPU. For reduce the comsupmtion WIFI must be put in sleep. This reduce the consumption aprox by 33% to from 8.1mA to 5.5mA. I haven't note any diference between WIFI_LIGHT_SLEEP or WIFI_MODEM_SLEEP. If no delay() o delay(0) WIFI doesn't sleep.
I will test the Espurna with WIFI_MODEM_SLEEP to see how it works. 
The Sonof T1 I use for this test have a current comsumption with only power suply, no CPU board, of 3,7 mA. Then the CPU and WIFI in Sleep only adds 2 mA to the standby current of the power suply.
Note: current is for aparent power at 230 V, measured with tester.

WIFI_SLEEP_MODE | LOOP_DELAY_TIME | mA
-- | -- | --
WIFI_NONE_SLEEP | 250 | 8,1
WIFI_NONE_SLEEP | 10 | 8,1
WIFI_NONE_SLEEP | 5 | 8,1
WIFI_NONE_SLEEP | 1 | 8,1
WIFI_NONE_SLEEP | 0 | 8,3
WIFI_MODEM_SLEEP | 1 | 5,5
WIFI_MODEM_SLEEP | 0 | 8,3
WIFI_LIGHT_SLEEP | 250 | 5,5
WIFI_LIGHT_SLEEP | 50 | 5,5
WIFI_LIGHT_SLEEP | 1 | 5,5
WIFI_LIGHT_SLEEP | 0 | 8,3
only power   suply Sonoff T1 | 3,7

mcspr, 2019-02-16:
@oscarsan1 correct, delay simply sets up software timer to return back from system to the app (loop) function after specified time:
https://github.com/esp8266/Arduino/blob/e721089e601985e633641ab7323f81a84ea0cd1b/cores/esp8266/core_esp8266_wiring.cpp#L41-L57
nonos sdk blob still does it's thing

also, see https://blog.voneicken.com/2018/lp-wifi-esp8266-assoc/

xoseperez, 2019-02-21:
@oscarsan1 so your findings suggest that using a `LOOP_DELAY_TIME` of 1 ms does make some (small) difference but using bigger times do not. Would you agree on reducing the default value from 10 to 1ms?

oscarsan1, 2019-02-21:
Hi. Regards to comsumption is the same 10ms or 1ms. I think that sleep() has no sense other than put the WIFI to sleep, and this happens if delay() is bigger than "0", delay(1) is enough. But WIFI_SLEEP_MODE has to be WIFI_MODEM_SLEEP or WIFI_LIGHT_SLEEP.
 
I'm testing in several Sonoff T1 and in Arilux LC02 with LOOP_DELAY_TIME=1 and WIFI_SLEEP_MODE=WIFI_MODEM_SLEEP without problem at the moment.

xoseperez, 2019-02-22:
OK, I'm removing the delay if `WIFI_SLEEP_MODE` is `WIFI_NONE_SLEEP` and reducing it to 1ms by default for the other cases.

closed 2019-02-22 by xoseperez

oscarsan1, 2019-02-22:
Sorry for the confuion, I mean that the delay must be greater than 0. 1 is ok. Dont put the delay(0). 
Then if WIFI_SLEEP_MODE=WIFI_MODEM_SLEEP the WIFI is put in sleep mode when not in use, saving 30% of energy. WIFI wake up automaticaly when a message arrive.

mcspr, 2019-02-25:
@oscarsan1 btw have you tried measuring consumption when setting cpu speed to 160MHz? 
Arduino IDE allows to change it in the menu.
PlatformIO requires additional env setting `board_build.f_cpu = 160000000L` (e.g. find your environment in platformio.ini and add it there)

Note that current `dev` has broken loop delay, checkout [`64b5155`](https://github.com/xoseperez/espurna/tree/64b5155aa7d4f28f6b4f6dc57d6cc6b560edd5d8) instead (also see #1574)

oscarsan1, 2019-02-25:
I haven't seen any diference in current consumption with 160 MHz compared with 80 MHz. Same comsumption!!!. I test it with WIFI_MODEM_SLEEP and WIFI_NONE_SLEEP and whith delay(0) and delay (1). 

Xose, I think the current implementation of this isue adds some complexity in the code. I think that the delay(1) is good also for  WIFI_NONE_SLEEP, also reduce consumption by near 5%. 

My recomendation is to use WIFI_MODEM_SLEEP and delay(1). I test it for a week with several sonoff T1 and Airlux LC10 without problem. I test only web acces, not yet MQTT. If WIFI_NONE_SLEEP is needed in any case, delay(1) doesn't disturb too much and reduce consumption by near 5%.