xoseperez/espurna #1543: cannot read bme280

fragolinux, 2019-02-10:
trying to read a bme280 (sda on gpio0, scl on gpio2) on wemos d1 mini using espurna-1.13.3-wemos-d1mini.bin but with no success... i don't understand if i2c support is enabled or not in this binary, as i can give these commands and work:

set BMX280_ADDRESS 0x76
set i2cSDA 0
set i2cSCL 2

i took the address via i2cscan flashing tasmota, as giving "i2c.scan" via usb serial console or debug reports: -ERROR unknown command

a little help, please? Thanks
how to know which features are enabled in each precompiled binary firmware?

```
[222313] [MAIN] ESPURNA 1.13.3 (e2641b9)
[222313] [MAIN] xose.perez@gmail.com
[222314] [MAIN] http://tinkerman.cat

[222314] [MAIN] CPU chip ID: 0xC76077
[222318] [MAIN] CPU frequency: 80 MHz
[222321] [MAIN] SDK version: 1.5.3(aec24ac9)
[222325] [MAIN] Core version: 2.3.0
[222328] [MAIN] Core revision: 159542381
[222331] 
[222333] [MAIN] Flash chip ID: 0x1640EF
[222336] [MAIN] Flash speed: 40000000 Hz
[222339] [MAIN] Flash mode: DOUT
[222342] 
[222344] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[222351] [MAIN] Flash size (SDK)    :  4194304 bytes / 1024 sectors (   0 to 1023)
[222358] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[222365] [MAIN] Firmware size       :   463392 bytes /  114 sectors (   1 to  114)
[222372] [MAIN] Max OTA size        :  2674688 bytes /  653 sectors ( 115 to  767)
[222381] [MAIN] SPIFFS size         :  1028096 bytes /  251 sectors ( 768 to 1018)
[222386] [MAIN] EEPROM size         :     4096 bytes /    1 sectors (1019 to 1019)
[222394] [MAIN] Reserved            :    16384 bytes /    4 sectors (1020 to 1023)
[222401] 
[222402] [MAIN] EEPROM sectors: 1019, 1018, 1017, 1016
[222406] [MAIN] EEPROM current: 1019
[222409] 
[222411] [MAIN] EEPROM:  4096 bytes initially |   181 bytes used ( 4%) |  3915 bytes free (95%)
[222420] [MAIN] Heap  : 36336 bytes initially | 11920 bytes used (32%) | 24416 bytes free (67%)
[222427] [MAIN] Stack :  4096 bytes initially |  1256 bytes used (30%) |  2840 bytes free (69%)
[222435] 
[222436] [MAIN] Boot version: 6
[222439] [MAIN] Boot mode: 1
[222442] [MAIN] Last reset reason: External System
[222446] [MAIN] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[222459] 
[222460] [MAIN] Board: WEMOS_D1_MINI
[222464] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[222479] [MAIN] WebUI image: SMALL
[222482] 
[222483] [MAIN] Firmware MD5: 490379768d58b56ae182e4243a31534c
[222489] [MAIN] Power: 2714 mV
[222491] [MAIN] Power saving delay value: 10 ms
[222495] 

```

xoseperez, 2019-02-11:
As you can see in the "Support" like of the dump, I2C is not enabled, also there is no support in the binary for BME280 sensor... You should be building a custom firmware for the WEMOS_D1_MINI with BMX280_SUPPORT setting to 1. Check then `arduino.h` file for easy cofiguration.

Each binary is target to a specific device, for general purpose devices like a Wemos D1 custom builds are required (except if you are using a relay shield, for instance, because there is already an image for that).

fragolinux, 2019-02-11:
oh, thanks, i didn't know that... i'll go recompiling, then
is there a list of what features are enabled in each binary firmware?

Valcob, 2019-02-11:
@fragolinux yup check the hardware.h there are board definitions per each device

closed 2019-02-11 by fragolinux