xoseperez/espurna #1544: [PLUGIN] iRobot Roomba

Danny89530, 2019-02-11:
Hi @xoseperez,
In the past two weeks I've worked on a PlugIn that add the ability to use your (great) ESPurna with the Roomba vacuum cleaner.
I'm not a Arduino guru and I'm sure there are a lot of things that can be simplifyed in the code or (maybe) needs to be rewritten because of my poor knowledge.
I've used the RoombaArduino Library available at https://github.com/Apocrathia/Roomba.

Actually the PlugIn has no sense if not used with MQTT support and it accepts the below commands:
```
Cleaning Functions:
	- Clean
	- Spot clean
	- Dock
	- OFF (Sleep/StandBy)
	- Pause
	- Resume
iRobot Functions:
	- NtpSync
	- Set Time		--> Format: DoW,HH,MM
	- Set Schedule		--> Format: DoW1,HH1,MM1;DoW2,HH2,MM2;... or clean

Data returned from MQTT:
	- Battery Pct		0-100
	- Battery Temp		celsius
	- Bin Full 		0/1			--> Not Available at the moment..
	- Bin Present		0/1			--> Not Available at the moment..
	- Charging State	<String Values>
	- Coordinates		X, Y, Theta
	- Mode			<String Values>
	- State			<String Values>
```

At the time of writing, the code (apparently) works fine, updating the RoombaState in the expected conditions, I'm having just a little bit of trouble with X, Y coordinates.
Actually the distance travelled is reported in millimeters, so the X, Y coords have a *big range* of value.
I'm working on a solution where, converting the distance in centimeters, we don't loose the travelled distance in approssimation/changing of direction/etc..

In the zip file ([iRobot Plugin.zip](https://github.com/xoseperez/espurna/files/2850582/iRobot.Plugin.zip)) there is also the official documentation of OpenInterface (provided by iRobot) where, in the third page, is explained how to connect Arduino (or a NodeMCU, in my case) to the Mini-DIN Connector.
The PDF is specifically for 600 models but it is fully compatible with other moldels (I've tested it on a 886/782 models with success).

I hope you enjoy this and help me to obtain a fully funcional plugin :)

Regards.
Danilo

AllWize, 2019-02-11:
Cool. I have a Roomba 500 myself and I also hacked some time ago. I will be great to have it monitored using ESPurna. I will give it a look.

xoseperez, 2019-02-11:
Great!
I have a Roomba 500 series that I hacked some time ago. It will be great to have it working with ESPurna. Will give it a look!

Danny89530, 2019-02-16:
Hi @xoseperez,
I've updated the zip file. Just a litte bit of bugfixes.
--> [iRobot Plugin.zip](https://github.com/xoseperez/espurna/files/2871860/iRobot.Plugin.zip)

Regards,
Danilo

Danny89530, 2019-02-20:
Hi @xoseperez,
I've updated the zip file again. Just a litte bit of bugfixes.
--> [iRobot Plugin.zip](https://github.com/xoseperez/espurna/files/2884249/iRobot.Plugin.zip)

Regards,
Danilo

Danny89530, 2019-02-25:
Other bugfixes..
--> [iRobot Plugin.zip](https://github.com/xoseperez/espurna/files/2901522/iRobot.Plugin.zip)

Danny89530, 2019-02-28:
Updated for 1.13.5 compatibility but with the old known bugs ^^
--> [iRobot Plugin.zip](https://github.com/xoseperez/espurna/files/2914318/iRobot.Plugin.zip)

Regards,
Danilo

Danny89530, 2019-03-09:
A lot of bugfixes..
--> [iRobot Plugin.zip](https://github.com/xoseperez/espurna/files/2948851/iRobot.Plugin.zip)