xoseperez/espurna #1546: Publish MQTT state regardless of current state

skorokithakis, 2019-02-12:
I am using the MQTT API to toggle relays. If try to turn a relay on but it's already on, no state is published. This is generally fine, but causes a problem with some apps that expect state.

Since there's no harm in publishing state, I would like to request that state is published on any command, regardless of whether that command ends up changing state or not.

Thanks!

wsw70, 2019-02-14:
Out of curiosity - do you have any scenarios in mind which would require this functionality?

I heavily rely on the state when using Home Assistant and so far the state was a combination of device feedback (when its state change) and retained messages (for the initial state when HA starts). Now that I think on it I may be missing some edge cases in my setup, though.

skorokithakis, 2019-02-14:
Yes, it's mostly for stateless applications who wait for espurna in order to learn about the state, but that only happens when their initial "guess" is correct. Retained messages would have also been great, but I couldn't find a way to retain the state message.

wsw70, 2019-02-14:
Hmmm, I had a closer look at the messages because I was (and am) a bit confused now.

I have two retained messages per device: 

- `<root>/relay/#` which is `0` or `1` depending on the state (on/off)
- `<root>/status` which is a heartbeat updated every few minutes with `1` when the switch is up, then `0` when it disconnects from the broker

So I do have a retained message with the current state.

What confuses me is that the MQTT tab on the device does not mention the state message:

> - `<root>/relay/#/set` Send a 0 or a 1 as a payload to this topic to switch it on or off. You can also send a 2 to toggle its current state. Replace # with the switch ID (starting from 0). If the board has only one switch it will be 0.
> - `<root>/status` The device will report a 1 to this topic every few minutes. Upon MQTT disconnecting this will be set to 0.

In practical terms, when I sent a `0` or `1` to `<root>/relay/#/set`, I get back a retained `0` or `1` on  `<root>/relay/#`

skorokithakis, 2019-02-14:
I see this behaviour as well. That's a better solution, except the topic I'm interested in (`root/data`) is missing (not retained). If that could be retained as well, it would solve my problem.

mcspr, 2019-02-15:
See #1503 regarding retain flag for json output

`<root>/data` is used for everything. Simply retaining last message with `"relay/0": 0` would not work with multiple relay boards, as you would not know any other relay status. IMO what could be done specifically for relays is to split things into a different topic, like `<root>/relay/status -> [0,0,0,0]` and retain that.
Maybe something like that for the device status too, so it does not repeat things like "mac", "ip" etc.

skorokithakis, 2019-02-16:
Ah, yes, that makes sense... As you say, maybe a retained device status key?