xoseperez/espurna #1549: GU10 Lamp - Similar to AI-Thinker

jkalousek, 2019-02-13:
I have got this GU10 led lamp from [Ebay ](https://www.ebay.com/itm/1-10PC-GU10-RGB-Smart-Bulb-Wireless-WiFi-App-Remote-Ctrl-Light-for-Alexa-Google/173724116351).

I assumed that it will be similar to AI-Thinker lamps as it uses same mobile app etc.
Board is in fact very similar, well half of it...

I tried to flash espurna-1.13.3-aithinker-ai-light.bin but no matter what I do in the web interface light is still at maximum brightness.

[image]

Chip at the top left is SLM211A (LED PWM driver) which is controlling white channel on GPIO4.
It is using GPIO0, GPIO2 & GPIO15 for RGB, I cannot say exactly which one is which but I'm going to figure it out.

I will add more info.

albyg, 2019-02-28:
I have this same bulb from Tuya. I flashed the ESP with Tasmota 6.4.1
Then...
1. Defined as Module Type "Generic" 
2. Define GPIO14 (R) as PWM1
3. Define GPIO012 (G) as PWM2
4. Define, GPIO13 (B) as PWM3
5. Define GPIO04 (W) as PWM4
6. SetOption15 1
7. SetOption30 0

It will then operate as a light with fully adjustable colour and white light. You can either use the web interface (IP address) or MQTT to operate.

xoseperez, 2019-03-17:
@albyg, @jkalousek says it has a LED driver on board, it doesn't make sense that it can be directly controller using PWM from the microcontroller....

albyg, 2019-03-18:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/xoseperez/espurna/issues/1549#issuecomment-473713538",
> "url": "https://github.com/xoseperez/espurna/issues/1549#issuecomment-473713538",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

The SLM211a is exactly that. A LED driver that uses PWM directly from the MCU. It also limits the total current to the LEDs by use of a setting resistor on one of the input pins.RegardsAlbert Girardi

Wferr, 2019-03-22:
I just got a light bulb that uses the same control board and am wondering what firmware to use on it? Thanks.

ColinShorts, 2019-03-22:
There are no suitable devices currently defined in espurna that will work with this device, so someone will need to test and add it via pull request.

I'd suggest using something like the definitions for the  IWOOLE_LED_TABLE_LAMP (RGBW over pwm) as a base to start on.

Ignoring my accidental change to Readme.md and poor git skills, take a look at PR #1198 of you're unsure what should be changed.

albyg, 2019-03-22:
Tasmota will support this board with set up mentioned earlier.

ColinShorts, 2019-03-22:
@albyg - based on the gpio pinout above I've built the attached firmware (unimaginatively called "generic gu10".

@Wferr - if your bulb is based on the exact same board then it might work for you too.

Please test if you can and confirm/deny if any changes need to be made, I could probably do a PR for you.

[firmware.bin.zip](https://github.com/xoseperez/espurna/files/2996742/firmware.bin.zip)

Wferr, 2019-03-22:
@ColinShorts Just flashed and working perfectly! RGB is correct and white channel works! Thanks!

ColinShorts, 2019-03-22:
No worries, I'll submit a pull request sometime tomorrow. Do you have a manufacturer and device name for this bulb/controller?

Wferr, 2019-03-22:
Don't really have a name for the bulb or controller, the aliexpress link where I bought it from is above, its all super generic.

ColinShorts, 2019-03-22:
I was starting to think a naming convention based on purpose and gpio assignment might make sense, but I'd bow to @xoseperez and @mcspr for guidance on that. Version 2.0 will likely resolve most of these device configuration issues, but I don't know how far off it is.

closed 2019-03-25 by xoseperez

TimelessNL, 2019-08-25:
@jkalousek may I ask how you removed the plastic top cover from this GU10 style bulb? Is it easily removable without much damage to the bulb?

TimelessNL, 2019-10-09:
To answer my own question, there is no need to remove the cover. These can now be flashed with [tuya-convert](https://github.com/ct-Open-Source/tuya-convert)

Wferr, 2019-12-02:
With the espurna 1.14.0 release I cant update my light that used the generic GU10 bulb without getting "There was an error trying to upload the new image, please try again (ERROR 8)." Which I think is due to a issue with [flash size #433](#433) but I don't know.

mcspr, 2019-12-03:
@Wferr It is. [Error codes are from the Core](https://github.com/esp8266/Arduino/blob/074a8016bfafc23c52368d5cca6880b36f102bc0/cores/esp8266/Updater.h#L17), device flash config is set to >1MB.
Same as https://github.com/xoseperez/espurna/issues/1959, atm only way is to re-flash .bin with correct values.

Wferr, 2019-12-03:
@mcspr Is there no way to OTA a flash to correct? I have to reconnect the light to a serial adapter and flash?

mcspr, 2019-12-04:
@Wferr Yes. Updater class won't allow OTA and there is no mechanism to avoid that check. Just make sure to either specify 1MB size or, if using esptoo.py, allow it to autodetect that value by using `--flash_size detect` or skipping that switch altogether.

PR #1650 (2019-03-25): Adding support for Generic GU10 bulb
This might also support more than just GU10 devices, but resolves #1549.