xoseperez/espurna #1550: [question/feature/plugin] reste when network not available

wsw70, 2019-02-14:
I have a need which is probably too specific to be included as part of the firmware: I use (among others)  a Sonoff Basic behind two devices:

- a server
- a set top box provided by the ISP (Internet / TV / phone)

The idea is that when things go horribly wrong, a forced power reset of the devices can be the ultimate save. This is true for the server (it happened that I did some things remotely which cut me off and rebooting the server brought it back to a working state), and certainly for the set top box which is a big pile of [censored], provided by the largest French national fiber provider whom I would like to thank for the reliability of their equipment /s

Today I have a script which runs periodically and check whether Internet (`1.1.1.1`) is available - if not then after a few tries an MQTT message is sent which switches off the switch, waits a moment and switches it back. This is great until the machine which runs the script is not the one which is faulty (or the network, or the broker or anything else on the way).

This is why I am considering to try to write a plugin which would bring this functionality right into the switch.

- `considering` because the last C program I wrote was 20 years ago
- `plugin` because of the probable specificity - but I do not know whether plugins are supported (I did not see that in the docs, but has not extensively searched either).

This post is more a quest for information (and ideally documentation) than anything else (well except if someone did not realize they had the need and are jumping on the idea :))

mcspr, 2019-03-11:
I apologize for the delay in replying :)

There two parts to this:
something to trigger data collection, probably with sensor interface.
and something to act upon it.

First part can be based on lwip code (networking library, implementing TCP/IP stack). IDK though whether this approach is compatible with the old lwip1.4/core 2.3.0. Due to some timing issues in SDK and esp8266 missing sending ARP, there was a method proposed to make esp ping gateway's ip every now and then:
https://github.com/d-a-v/EspGoodies/blob/master/examples/pingAlive/pingAlive.ino
https://github.com/d-a-v/EspGoodies/blob/master/src/utility/ping.c

Second part can be achieved via #1521 , linking the sensor data into some flow

mcspr, 2019-08-01:
In the process of adapting ArduinoJson6, I needed to have some dummy sensor that does not need any hardware but still allows to report real (and changing) data. Following my own idea from the message above, I got a working sensor class that periodically does ping.

That's about it, because I am not sure what exactly should be it's output:
- keep internal state and do 5 pings. If at least one succeeded, sensor value is 'True' / 1. 'False' otherwise, repeating read cycle (this is what i have implemented, following pingalive idea)
- total counts for successes / failures? like events sensor "count" magnitude or cli ping utility
- force instantaneous ping result on every read. meaning, we would need to block execution and wait until ping result comes back. 1 second timeout, for example. and only 1 ping would happen

amarendrapramanik, 2024-05-08:
The issue raised by @wsw70 has a practical sense.

Scenario I: I switch on some appliances using any of the integration like Home Assistant, Alexa etc. and suddenly the network gone. In this scenario we do not have access to any of the integration and as a result the appliances will be ON state till the point the network again restores.

Scenario II: Similar to scenario I but in this case, router connected without internet as a result the integration which are mostly dependents on the internet connectivity won't work till the point again the connectivity restored.

Is there any possibility to implement a rescue mode for all the relays after a configurable delay, that will make the relays in OFF mode.

Request you to please think and if possible, to mark this enhancement into the next release.

mcspr, 2024-05-09:
Setting pulse timer works here? Timer will continuously tick down to 0 while the relay is ON, and periodically publishing `my-switch-device/relay/0/set 1` will reset it back the initial value. When this does not happen (publisher cannot reach the broker, or device cannot read the broker data), relay would eventually turn off.

> Scenario I: I switch on some appliances using any of the integration like Home Assistant, Alexa etc. and suddenly the network gone. In this scenario we do not have access to any of the integration and as a result the appliances will be ON state till the point the network again restores.

For MQTT specifically, this was implemented as `mqttDisconnectReaction#` (https://github.com/xoseperez/espurna/pull/653) where `#` means Nth relay action. Right now it is called `relayMqttDisc#`. Usually, time between MQTT disconnect and device acting upon it is controlled by MQTT keep-alive interval. Although, normal disconnection also triggers, current clients do not distinguish between timeouts and normal disconnect requests. Or, on reconnect, for that matter.
```console
set relayMqttDisc0 off
```
(or `relayMqttDisc1` for relay 1, and etc.)

> Scenario II: Similar to scenario I but in this case, router connected without internet as a result the integration which are mostly dependents on the internet connectivity won't work till the point again the connectivity restored.

Scripting can be used? Although it lacks some obviously useful operators, as it turns out
https://github.com/xoseperez/espurna/wiki/RPN-Rules#wifi - `rssi` output is equal to -127 when disconnected; maybe `$connected` or `connected` op can be added for simplicitly
https://github.com/xoseperez/espurna/wiki/RPN-Rules#relays - `$relay0` variable for status, `0 0 relay` to turn specific relay off.

https://github.com/xoseperez/espurna/wiki/RPN-Rules#mqtt-variables also might help. Note that *any* type of variable works (gtg edit the wiki)

amarendrapramanik, 2024-06-23:
I verified the behavior described above, but unfortunately rssi does not goes to a value -127 when disconnected from internet and which is quite logical. The main reason is that router is still connected with the home network but unfortunately it loses the internet connectivity. Moreover, as you said the implementation with mqttDisconnectReaction# for the rescue mode operation is not very firm idea because I might not use MQTT at all.

Is it possible to implement an even in the **wifi** class itself which notify when the router loses/ restore connectivity from the external world in addition to the other events it implemented. If anyone wants to subscribe to the event, they can subscribe and implement rescue mode operation according to their choice. In my case I should be switch off all the relays if connectivity does not return back for may my 300 sec.

mcspr, 2024-06-24:
Returning to the OT then - what is a definitive way to do that? Ping is one, yes. Other way is to poll connectivity services like iOS, Windows, Android, etc., `http //www.google.com/generate_204` and others. So, it probably should be several checks that determine the need for a rescue state?

It would not be part of WiFi, which is purely a physical layer here.

If device is accessible externally, does the pulse suggestion above work? Until something stops pulse requests, device would stay on. Pulse time becomes such timeout.