xoseperez/espurna #1552: Unsynced group feature

skorokithakis, 2019-02-16:
It would be nice if espurna could listen on some keys where it would get commands that would be shared with other instances. My use case is having a group called "house" where I could send commands for all switches in the house, such as "turn all off" or "toggle all". Currently there is no way to do this without sending one command per relay,  as far as I know.

mcspr, 2019-02-17:
Does the mqtt group feature cover it? https://github.com/xoseperez/espurna/wiki/MQTT#relay--color-synchronization-across-devices
You can define arbitrary topic per switch that accepts same payloads as `/set`: 0 for off, 1 for on and 2 for toggle. Using the same one across multiple devices allows to control them via single topic. Note that just as the status topics, it will not set state from retained message when connecting (but probably should...)

cc @abmantis from https://github.com/xoseperez/espurna/issues/145#issuecomment-464078322

ColinShorts, 2019-02-17:
I don't think the current group setting would work in this context as they all have the same state. I use groups to control my hall lighting over three floors. i.e: I can switch off lights downstairs and upstairs from the middle floor (sonoff t1 2ch's), the middle floor and downstairs can be controlled from another T1 2ch at the bottom of the stairs, and the same from the top floor.

I'd be more inclined to handle this logic in OpenHAB (or other) by grouping the lights there. You can setup a virtual switch that will accept ON, OFF, and TOGGLE, and that would cover the use case presented by @skorokithakis

skorokithakis, 2019-02-17:
Yes, as Colin said, the current group functionality doesn't work as it would sync all the lights. Grouping everything via OpenHAB doesn't work so well for me because I don't have OpenHAB, or anything else on top of the switches.

mcspr, 2019-02-17:
So, do you mean switch should not publish state to such topic? Only listen.

skorokithakis, 2019-02-17:
Yes, exactly.

abmantis, 2019-02-17:
As I've said in https://github.com/xoseperez/espurna/issues/145#issuecomment-464078322 this is useful. 
I also think that espurna would only need to subscribe to the group topics, and never publish there.

ColinShorts, 2019-02-17:
@skorokithakis, what are you using on the other side of the mqtt server to control the lights? Would an acl on the mqtt topics as mentioned in #145 work for you? I'd imagine the espurna devices would be prevented from publishing to the topic, whereas the "remote" would be the only thing that was allowed to publish.
Are you using mosquito?

skorokithakis, 2019-02-17:
@ColinShorts I am, but I don't see how this could be applied... Are you saying I should use a sync group topic and prevent the devices from posting to it with an ACL, making the sync group read-only?

ColinShorts, 2019-02-17:
I believe that might be the simplest option (without code changes in espurna). It's not something I've tried yet, but I don't see why it wouldn't work unless espurna has issues if it can't write to the group topic.

skorokithakis, 2019-02-17:
Seems like a reasonable workaround, thanks!

ColinShorts, 2019-02-17:
Please post your findings, it could help others. I don't mind giving it a try if you hit any snags.

skorokithakis, 2019-02-17:
I will if I try it, I currently don't authenticate any devices so it might be too much of a hassle over just having my script post to all the topics. If I do try, I'll definitely report back.

mcspr, 2019-02-18:
I think this can be easily added as 3rd group sync mode: "Receive only"

Mosquitto ACL rules should work too, as far as I know. It will silently drop messages.

skorokithakis, 2019-02-18:
Actually, I think that's a great idea.

mcspr, 2019-02-21:
See #1561 (not tested, but I think it should build)

ColinShorts, 2019-02-21:
Thanks @mcspr! @skorokithakis, can you test this by applying the patch and building the binaries you need? If not, what devices will you be testing with?

skorokithakis, 2019-02-21:
Sure, I can test this! I'm afraid it might take some time before I can get home, but I can try it when I do.

PR #1561 (2019-02-25): MQTT relay group: receive-only mode
Resolve #1552 

- introduce 3rd way that mqtt groups work - only receive messages, do not send anything to the group topic
- rename mqttGroupInv -> mqttGroupMode
- sync mode type tokens
closed 2019-02-25 by xoseperez