xoseperez/espurna #1555: Hostname config should go to WIFI config options

brunetton, 2019-02-18:
Hi,

## Quick story

It took me a while to find the Hostname config option that is actually in `General` UI config page. I would have found it more intuitive to find it in `WIFI` page.

## Long story

After configuring my second device, I wanted to change hostname (for security reasons). So I went to WIFI config page but didn't find it, so I hex-edited bin file and replaced ESPURNA by another 7 chars string. This is quick and very dirty, of course (I hadn't enough time to compile from sources and have a look in config file).

After a while of use I finally seen it in General config page and was finally happy :) But maybe newcomers like me missed it also.

xoseperez, 2019-02-21:
I see what you mean, but the hostname is also used in the MQTT tab (as a topic placeholder) or in the Alexa feature (to name the switches after it), so it's not only WiFi related.

brunetton, 2019-02-21:
OK, in this case I understand. Maybe a little tip text can indicate that hostname can be configured at this place could be added?

xoseperez, 2019-02-21:
Done :+1:

closed 2019-02-21 by xoseperez