xoseperez/espurna #1558: Hotspot again

eduardocanton, 2019-02-18:
Hello,

I have several sonoff basic and I uploaded the firmware 1.13.2 (binnary file)

The first time, I conected with sonoff hotspot and I configured the sonoff with my home wifi and works!.

The problem is when I turn off and turn on the sonoff again, then appears the wifi again (ESPURNA_XXX) and not conect to sonoff by wifi and I cannt connect to ESPURNA_XXX either.

which can be the problem?

thanks

ColinShorts, 2019-02-18:
The admin password becomes the wifi password, have you tried that?

eduardocanton, 2019-02-19:
yes, the first time, when i connect i change the admin pass and the wifi pass to connect. i connect to my home wifi and works with the new ip.  the problem is if I turn off and on the espurna again, then doesnt connect to my wifi network and it´s appear the EspurnaXXXX hotspot wifi generated.

thank in advance

ColinShorts, 2019-02-19:
If the password hasn't returned to the default, "fibonacci", you'll probably find it easier to flash the units again. I think the latest firmware is v1.13.3, do you still see the same issue with that?

alternatively, you might be able to see what is happening if you disconnect from mains power and attach to the serial console. Try the "config" command, which should show all of the settings.

xoseperez, 2019-02-21:
When you connect again to the device in hotspot mode, can you check if it has stored the values in it's memory? Just go to the `DEBUG` tab and type `keys` to get all key-value pairs stored.

closed 2019-04-29 by stale[bot]