xoseperez/espurna #1560: mqtt unable to connect causes reboot

wired-dev, 2019-02-21:
on the latest dev branch; I'm having issues with the bulb reseting if mqtt is enabled and the bulb is unable to connect. once I disabled it it became stable.

mcspr, 2019-02-21:
@wired-dev with mqtt server set to non-existent address i could reporoduce this on current dev
can you verify fix in #1563?

wired-dev, 2019-02-21:
I applies the patch; and it does appear to resolve the issue.

0anton, 2019-02-21:
Could it be also related to my case: all my sonoff switches start flapping if the wifi is not available. Switches have also MQTT configured and obviously the MQTT server address becomes invalid as soon as the network disappears. Badly, once the router is itself on the sonoff switch, switch flapping never allow the router to boot and switch never find the network again...

How can I apply the fix to my switches?

wired-dev, 2019-02-21:
yes 0anton; I do think this is your issue. need to build it from source and apply the mod the your mqtt.ino file

0anton, 2019-02-21:
May be Max @mcspr can integrate the fix into dev code line since wired-dev has already tested and confirmed the fix is working? I don't have dev environment unfortunately for ESPxxx to provide the second test. Already running switches on latest 1.13.3 build.

closed 2019-02-21 by xoseperez

0anton, 2019-02-21:
Hi Xose @xoseperez, thank you for merging Max patch! That was fast! Do you have an automatic build of the dev codeline somewhere? I could test it, if I get the image.

mcspr, 2019-02-21:
@0anton see https://github.com/mcspr/espurna-nightly-builder/releases , it should build another one in about ~7 hours from now.

0anton, 2019-02-22:
Thank you, @mcspr Max! I uploaded dev build to 3 sonoff POW switches and looking for their behavior.

PR #1563 (2019-02-21): Wait until mqtt client has finished trying to connect
fix #1560 
do not trust mqtt client to actually connect, lock when connecting and wait until unlocked