xoseperez/espurna #1564: Transition time is ignored on Home Assistant

nerdosity, 2019-02-21:
**Bug description**
Transition time in Home Assistant is ignored, but transition is marked as supported.
This is very useful since you can turn on, then dim a light in 30 minutes, and just turn it on again if there is some movement. It just require one automation, and no control on dim state, brightness etc.

**Steps to reproduce**
Just try to dim (turning on or off) a led strip or a lamp with transition time. It is ignored, it turns off istantaneously.

**Expected behavior**
I have another LED (WS8211) in another ESP8266 installation made via ESPHomeLib. Transition there is not ignored, it actually dim in the given seconds.

**Screenshots**
[image]

**Device information**

[177113] [MAIN] ESPURNA 1.13.3 (e2641b9)
[177114] [MAIN] xose.perez@gmail.com
[177115] [MAIN] http://tinkerman.cat

[177116] [MAIN] CPU chip ID: 0x1509EC
[177117] [MAIN] CPU frequency: 80 MHz
[177121] [MAIN] SDK version: 1.5.3(aec24ac9)
[177125] [MAIN] Core version: 2.3.0
[177128] [MAIN] Core revision: 159542381
[177131] 
[177132] [MAIN] Flash chip ID: 0x1440C8
[177137] [MAIN] Flash speed: 40000000 Hz
[177139] [MAIN] Flash mode: DOUT
[177142] 
[177143] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[177151] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[177161] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[177166] [MAIN] Firmware size       :   486848 bytes /  119 sectors (   1 to  119)
[177172] [MAIN] Max OTA size        :   536576 bytes /  131 sectors ( 120 to  250)
[177179] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[177186] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[177194] 
[177195] [MAIN] EEPROM sectors: 251, 250
[177198] [MAIN] EEPROM current: 251
[177202] 
[177203] [MAIN] EEPROM:  4096 bytes initially |  1249 bytes used (30%) |  2847 bytes free (69%)
[177213] [MAIN] Heap  : 37160 bytes initially | 21232 bytes used (57%) | 15928 bytes free (42%)
[177219] [MAIN] Stack :  4096 bytes initially |  1248 bytes used (30%) |  2848 bytes free (69%)
[177227] 
[177228] [MAIN] Boot version: 4
[177230] [MAIN] Boot mode: 1
[177233] [MAIN] Last reset reason: External System
[177238] [MAIN] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[177251] 
[177253] [MAIN] Board: AITHINKER_AI_LIGHT
[177256] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[177273] [MAIN] WebUI image: LIGHT
[177275] 
[177276] [MAIN] Firmware MD5: aa8f7f4facf64a36001ffc51ce1e30bc
[177281] [MAIN] Power: 3186 mV
[177283] [MAIN] Power saving delay value: 10 ms

AND

[208684] [MAIN] ESPURNA 1.13.3 (e2641b9)
[208685] [MAIN] xose.perez@gmail.com
[208686] [MAIN] http://tinkerman.cat

[208687] [MAIN] CPU chip ID: 0x060DA3
[208688] [MAIN] CPU frequency: 80 MHz
[208692] [MAIN] SDK version: 1.5.3(aec24ac9)
[208696] [MAIN] Core version: 2.3.0
[208699] [MAIN] Core revision: 159542381
[208702] 
[208704] [MAIN] Flash chip ID: 0x1640C8
[208708] [MAIN] Flash speed: 40000000 Hz
[208710] [MAIN] Flash mode: DOUT
[208713] 
[208714] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[208722] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[208729] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[208736] [MAIN] Firmware size       :   488096 bytes /  120 sectors (   1 to  120)
[208746] [MAIN] Max OTA size        :   532480 bytes /  130 sectors ( 121 to  250)
[208751] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[208757] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[208766] 
[208767] [MAIN] EEPROM sectors: 251, 250
[208769] [MAIN] EEPROM current: 251
[208772] 
[208773] [MAIN] EEPROM:  4096 bytes initially |  1329 bytes used (32%) |  2767 bytes free (67%)
[208782] [MAIN] Heap  : 36776 bytes initially | 21392 bytes used (58%) | 15384 bytes free (41%)
[208790] [MAIN] Stack :  4096 bytes initially |  1376 bytes used (33%) |  2720 bytes free (66%)
[208800] 
[208801] [MAIN] Boot version: 31
[208803] [MAIN] Boot mode: 1
[208804] [MAIN] Last reset reason: Power on
[208808] [MAIN] Last reset info: flag: 0
[208811] 
[208813] [MAIN] Board: ARILUX_AL_LC11
[208816] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[208831] [MAIN] WebUI image: LIGHT
[208834] 
[208835] [MAIN] Firmware MD5: 06a989f1f7eae7ebdfbbed8b353dd464
[208841] [MAIN] Power: 3280 mV
[208845] [MAIN] Power saving delay value: 10 ms

**Tools used**
(Docker on Synology) Home Assistant 0.87.0
(Package on Synology) Mosquitto 1.4-14-9

**Additional context**
*Add any other context about the problem here.*

mitchese, 2020-12-28:
If this is going to be implemented via MQTT, then this will need to go via the json payload rather than the default payload (current setup). Default does not support transition times:

https://www.home-assistant.io/integrations/light.mqtt/

mcspr, 2020-12-29:
Correct. Note that it also does not inherently depend on the MQTT JSON setting from our side, only one special topic would allow HA<->Espurna communication.
iirc @ngilles had some prototype of the module for the MQTT communication, and I am trying to find time to implement something from my side as well to avoid the current intertwined relay<->lights (i.e. you need relays to make lights ON/OFF work)

There's also an option to implement it through a custom API, wiring into HA directly via a python module
- https://github.com/home-assistant/core/tree/dev/homeassistant/components/tasmota
- https://github.com/home-assistant/core/tree/dev/homeassistant/components/esphome
but idk how I feel about that

commit 4d157ccd (2021-01-25): ha: discovery refactoring

- retain setting
- generate topic and message in the discovery object instead of
creating them all at once
- re-implement relays, lights and sensors discovery
- rework name normalization, use relative magnitude indexes
- replace original lights code with schema: json, subscribe to a special
topic to handle lights instead of using the common API
(still wip - colors, hs, temperature settings)

resolve #630
resolve #1564
resolve #2403
closed 2021-01-25 by mcspr