xoseperez/espurna #1565: InfluxDB doesn't build with latest Core

mcspr, 2019-02-22:
`Client` class happened to change between versions, ESPAsyncTCP's `SyncClient` is no longer compatible

Source: https://travis-ci.org/mcspr/espurna/jobs/496826731#L537
```
Building firmware images...
* espurna-1.13.4-dev-travis02.bin --- espurna/pwm.c: In function 'pwm_start':
espurna/pwm.c:394:13: warning: 'RTC_REG_WRITE' macro is deprecated [enabled by default]
RTC_REG_WRITE(FRC1_LOAD_ADDRESS, 0);
^
/home/travis/build/mcspr/espurna/code/espurna/influxdb.ino:15:12: error: cannot declare variable '_idb_client' to be of abstract type 'SyncClient'
SyncClient _idb_client;
^
In file included from /home/travis/build/mcspr/espurna/code/espurna/influxdb.ino:12:0:
.piolibdeps/ESPAsyncTCP/src/SyncClient.h:30:7: note:   because the following virtual functions are pure within 'SyncClient':
class SyncClient: public Client {
^
In file included from /home/travis/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClient.h:27:0,
from /home/travis/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from .piolibdeps/ESP Async WebServer/src/ESPAsyncWebServer.h:35,
from espurna/config/prototypes.h:167,
from espurna/config/all.h:35,
from /home/travis/build/mcspr/espurna/code/espurna/espurna.ino:22:
/home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:29:21: note: 	virtual int Client::connect(IPAddress&, uint16_t)
virtual int connect(CONST IPAddress& ip, uint16_t port) =0;
^
/home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:37:22: note: 	virtual bool Client::flush(unsigned int)
virtual bool flush(unsigned int maxWaitMs = 0) = 0;
^
/home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:38:22: note: 	virtual bool Client::stop(unsigned int)
virtual bool stop(unsigned int maxWaitMs = 0) = 0;
^
*** [.pioenvs/travis02/src/espurna.ino.cpp.o] Error 1
========================== [ERROR] Took 5.96 seconds ==========================
```

xoseperez, 2019-03-29:
This is related to https://github.com/me-no-dev/ESPAsyncTCP/issues/109.
Added a wrapper class in the meantime.

mcspr, 2019-03-30:
One interesting option is to use https://github.com/boblemaire/asyncHTTPrequest , but the library API needs some changing (one obvious case - double `String` copy instead of using pointers when using ::send())

closed 2019-09-30 by xoseperez

mcspr, 2019-09-30:
Also note that upstream has this fixed proper: https://github.com/me-no-dev/ESPAsyncTCP/commit/75c2513c57ac8415eab3670dba5c64e73090f5ee
Small fix required to have it working with ASYNC_TCP_SSL_ENABLED (should probably PR that)
https://github.com/xoseperez/espurna/pull/1909#issuecomment-533963474