xoseperez/espurna #1566: Weak RF signal from RF Bridge after flashing with ESPRUNA

eihnat, 2019-02-22:
Hi,

I managed to flash RF bridge (R2 v.1.0) with espruna. Also to setup codes with learn option. But issue i get is very weak RF signal (or as it seems).

When I try to switch switch to on or off, debug messages, seem to be ok, also red led flashes twice and it sends signal, but power plug won't pick up code. 

I have tried to move RF bridge very close (couple cm)to power plug and it works 100%. When having it approx 1 m or more, no change of status, seems power plug doesn't receive code.

From tests it seems working only to up to maybe 50 cm or less to be functional. 
Prior flashing to espruna, original eWelink firmware/software combination worked flawlesly at least for couple meters. I used it for several weeks without any signal message drop before. This started only after flashing.

Also, I have only flashed firmware on esp8266, not RF chip, nor did any HW modifications. I assume this is still ok process and should be fully functional.

Anybody experienced similar?
Thanks.

[596462] [WEBSOCKET] Requested action: relay
[596464] [RELAY] #0 scheduled ON in 0 ms
[596468] [RELAY] #0 set to ON
[596470] [RFBRIDGE] Enqueuing MESSAGE '244A0118038E575C0C' 4 time(s)
[597304] [RFBRIDGE] Action 0xA0
[597479] [RELAY] Setting relay mask: 1
[597807] [RFBRIDGE] Action 0xA0
[598596] [WEBSOCKET] Requested action: relay
[598598] [RELAY] #0 scheduled OFF in 0 ms
[598601] [RELAY] #0 set to OFF
[598604] [RFBRIDGE] Enqueuing MESSAGE '24400118038E575C03' 4 time(s)
[598801] [RFBRIDGE] Action 0xA0
[599437] [RFBRIDGE] Action 0xA0
[599612] [RELAY] Setting relay mask: 0
[599940] [RFBRIDGE] Action 0xA0
[600684] [MAIN] Uptime: 600 seconds
[600687] [MAIN] Heap  : 36120 bytes initially | 15984 bytes used (44%) | 20136 bytes free (55%)
[600689] [MAIN] Power: 3204 mV
[600690] [MAIN] Time: 2019-02-22 17:58:17
[600937] [RFBRIDGE] Action 0xA0

xoseperez, 2019-02-24:
The log looks OK, but 50cm is certainly not very far. I have not done extensive testing but I usually test my unit 3-4 meters away from the plug and I did some tests across walls and it could easily reach two rooms away...

As for the difference between the original firmware and ESPurna, you already know that all the RF front-end is managed by the secondary microcontroller. There is no documented code for TX power management so I can't think of a reason for this...

kiralikbeyin, 2019-02-26:
@eihnat 
1-Your device maybe broken double check antennas. 
2-Change your power supply.

closed 2019-05-04 by stale[bot]