xoseperez/espurna #1571: RF Bridge RF codes are not visible with 1.13.4

franki29, 2019-02-23:
Hi all,
after upgrade RF Bridge to new code, the RF codes are not visible at the web interface.
But still it is working

Br. Frank

mcspr, 2019-02-23:
WebUI from #1387 caused this. https://github.com/xoseperez/espurna/blob/2aa5c590666b5cf550d03cd832da4d5c36e83de1/code/espurna/rfbridge.ino#L107
I remember fixing the merge, but used the wrong variable! That line must send root object

PieBru, 2019-02-25:
wsSend(root) in line 107 of rfbridge.ino doesn't change the buggy behaviour.
Even setting the codes by hand in the RF page, after rebooting those codes disappear, but in the status page the switches goes on/off correctly, also corresponding MQTT messagges are sent correctly.
Thanks,
Piero

mcspr, 2019-02-25:
@PieBru #1578 should fix that

closed 2019-02-26 by xoseperez