xoseperez/espurna #1573: _debugSendSerial have hardcoded Serial

rafael-sanz, 2019-02-23:
Hi, _debugSendSerial have hardcoded Serial and must be DEBUG_PORT (case use of Serial1).

Fixed it, Serial1 came again to show the log but yet the next message is showed by Serial

[MDNSResponder] _parseQuery: Possible race-condition for host domain detected while probing.

regards

mcspr, 2019-02-24:
Thank you for noticing!
Additional note: affected 1.13.4 .bins are `huacanxing-h801` and `huacanxing-h802`

Second one does not come from ESPurna, but from newest Core release 2.5.0. See:
https://github.com/esp8266/Arduino/blob/2.5.0/libraries/ESP8266mDNS/src/LEAmDNS_Control.cpp#L191
https://github.com/esp8266/Arduino/blob/2.5.0/libraries/ESP8266mDNS/src/LEAmDNS_Control.cpp#L217
I'd create a new issue at https://github.com/esp8266/Arduino/issues about this (or straight up open a Pull Request, as it looks like leftover from the line above)

rafael-sanz, 2019-02-24:
Great news, thanks! ;-)

rafael-sanz, 2019-02-25:
Thanks! :+1:

closed 2019-02-25 by rafael-sanz