xoseperez/espurna #1574: Seriouse WiFi issue ESPURNA Version 1.13.4

AlbertWeterings, 2019-02-24:
After updating my device to 1.13.4 I found multiple device not coming online anymore. Debug via serial is not giving me any information. Devices that come online after the update do not have their web interface reachable after a few hours, they keep responding to MQTT commands.

On SonOff-basic after double pressing the button, serial debug says there will be a access point created but no access point comes online and if there comes one it drops out after a few seconds.

I flashed 1.13.3 in it and the device starts working again as it shout.
I flashed a later dev version and still WiFi is working correct.

I am using:
ESP8266 core 2.4.2 in all versions
Library JustWifi https://github.com/xoseperez/justwifi#2.0.2 in all versions

mcspr, 2019-02-24:
Thanks!

Can reproduce with both pre-built and 2.4.2. AP mode behaves erratically.
@AlbertWeterings for now, you can revert https://github.com/xoseperez/espurna/commit/31195ba3b62ff886313d0369baf4b6ab1a07187b locally. Not really sure what's halting WiFi operation. And I think it also affects connection, as the Basic I have sometimes cannot connect at all.

jnogues, 2019-02-24:
I have the same problem. 
Sonoff Basic flashed with *.bin file in the repositiry. Test with differents ssid and never connects.

AlbertWeterings, 2019-02-24:
If you go back to this state it is working all normal as @mcspr noted.
https://github.com/xoseperez/espurna/tree/31195ba3b62ff886313d0369baf4b6ab1a07187b

biddster, 2019-02-24:
Confirming wifi issues with 1.13.4. 

Sonoff RF updated via the web interface. Refused to connect to my home network and falling back to AP mode. If I connect to the AP, I'm unable to get to the web interface via either of the local addresses. 

I longed pressed the toggle button to reset and can see that the reset is successful. The AP is ESPURNA-xxxxxx. If I connect to that AP using the default creds, I'm still unable to get to the web interface via the local address.

I'll pop it out this evening and reflash to see what happens.

mcspr, 2019-02-25:
@AlbertWeterings what I meant is reverting to the state before that commit helped. i.e. [`64b5155`](https://github.com/xoseperez/espurna/tree/64b5155aa7d4f28f6b4f6dc57d6cc6b560edd5d8), where there is still `delay(10)` at the end of the `loop()` (which removal with the default builds seemingly broke things)

Some observations:
`WIFI_NONE_SLEEP` etc. are enums and not preprocessor defines, and always equal 0 in preprocessor context.
With default settings `WIFI_SLEEP_MODE == WIFI_NONE_SLEEP`, Basic connects / works as AP when CPU speed is set to 160MHz. platformio.ini env must have `board_build.f_cpu = 160000000L`, IDE has menu setting. 
Changing sleep mode to `WIFI_MODEM_SLEEP` does help too (note that `delay()` does not get included, because preprocessor statements are always false!)

So,  what I think needs to happen is to replace preprocessor checks of sleep mode with runtime setting like `wifiSleep` to allow changing it at runtime, so `loopDelay` can always be changed.

xoseperez, 2019-02-25:
I'm not noticing connectivity issues with my testing board and pre-built binaries, not with 2.4.2 binaries either. But I have seen the upgrade from web UI process fails most of the times... not sure yet if this could be related to the delay thing...

mcspr, 2019-02-25:
I had similar results on testing boards. Basic is behaving differently for some reason. Bad power? Badly drawn antenna?

xoseperez, 2019-02-25:
Basic has always had connectivity issues. I have even thought about stop supporting it (not really).

mcspr, 2019-02-25:
Web OTA likely fails because of progress messages sent via websocket get pinged back every time (`websock.send({})`). Delay may help there, but they can be safely removed. Progress is already tracked client-side. Or just replacing generic debug function with `DEBUG_PORT.printf_P(...)`

PieBru, 2019-02-25:
Similar behaviour on my RF Bridge R2, using just updated dev sources: **it authenticates and changes admin password, then it lands to a blank WEB page**.

Steps to reproduce inside VScode console on Arch Linux:

$ export ESPURNA_ENV="itead-sonoff-rfbridge-direct"
$ pio run -e $ESPURNA_ENV -t clean &&  pio run -e $ESPURNA_ENV && gulp
$ esptool --port /dev/ttyUSB0 --baud 115200 erase_flash
$ pio run -e $ESPURNA_ENV -t upload

It works in STA mode just after rebuilding with my wifi LAN creds in custom.h :
#pragma once
#define WIFI1_SSID                          "###"
#define WIFI1_PASS                          "###"

IMHO something messed up toghether with the 1.13.4 huge changes.
Thanks,
Piero

xoseperez, 2019-02-26:
It certainly looks like it depends on the board. I was having problems with a couple of Blitzwolf SHP2 but the other boards at home are working just fine. After @mcspr reverting the loopDelay issue everything works as normal.

I have removed the binaries from the 1.13.4 release. My idea is to release 1.13.5 with this fix as soon as possible (after some testing).

closed 2019-02-26 by xoseperez