xoseperez/espurna #1577: SonOff Pow R2 does not connect

decibelio99, 2019-02-25:
Hi there,
First of all, I'd like to thank your work, I have all my electrical gadgets connect with your firmware. Even, I have made a dry contact with a Sonoff basic that works perfectly.
Well, all less one, a sonoff Pow r2; every time I upload the firm, it works fine, but when I reboot for a second time, the pow becomes a brick. I have waitted for more than 3 minutes and I have tried with the 3 last releases of Espurna.
PS. The pow r2 works fine with Trasmota firm, but I prefer to use Espurna.
Thanks in advance

dreamfalcon, 2019-02-25:
Version 1.13.4?
If yes check this issue #1574

decibelio99, 2019-02-25:
I have the same problem with all versions, 1.13.2, 1.13.3 and 1.13.4

xoseperez, 2019-02-26:
Same problem since 1.13.2? Does it default to AP mode or do you have to flash it again?

decibelio99, 2019-02-27:
I've tried the three versions, but after setting up it, it didn't make anything, so I had to flash it to tasmota :(

xoseperez, 2019-03-17:
I have a POW R2 at home monitoring the washer machine and it works great... recovered fine after power cuts. I'm sorry I can't help without further info. If you ever give ESPurna a chance again please reopen if it keeps on failing.

closed 2019-03-17 by xoseperez