xoseperez/espurna #1581: SonOff pow R2 vs BLITZWOLF_BWSHPX vs BLITZWOLF_BWSHPX_V23

AlbertWeterings, 2019-02-25:
I noticed that different power energy meters are sending different sets of data via the MQTT json message /data

Some of these meters have the same data available in the web GUI so why not send the same data in the MQTT message. This makes it easier to handle if for example you want to store data in a database or external application.

I did a quick search to find where the data set is created but was not able to find it. If someone can point me in the right direction I would like to try to make sets for different meters the same.

AlbertWeterings, 2019-02-27:
I think I figured it out for SonOff Pow R2 but as i'm traveling I'm not able to check if this works. could someone put the file CSE7766Sensor.h in his sensors folder compile ESPURNA 1.13.4 or 1.13.5, write it to a SonOff pow R2 and check if:

1. ReactivePower is measured.
2. If it is in a MQTT channel.
3. If it is in the objects if JSON is enabled.
4. A list of all JSON objects.

[CSE7766Sensor.zip](https://github.com/xoseperez/espurna/files/2912211/CSE7766Sensor.zip)

AlbertWeterings, 2019-02-28:
Pull Request [#1591](https://github.com/xoseperez/espurna/pull/1591) solves my issue for CSE7766 Sensor based units.

xoseperez, 2019-03-17:
#1591 is mostly OK except for the reactive power calculation. There should really be a sqrt there :smile:

mcspr, 2019-03-21:
fixed by the b9029020a5f268e084f6a19ba7f56d73bb275c89

closed 2019-03-21 by mcspr