xoseperez/espurna #1583: espurna-wemos-d1-pzem004t binary

rradar, 2019-02-26:
Hello!

I wonder if it would be possible - as there are already a couple of custom binaries for the wemos d1 like:

- espurna-1.13.3-wemos-d1-tarpunashield.bin
- espurna-1.13.3-wemos-d1mini-relayshield.bin
- espurna-1.13.3-wemos-d1mini.bin

to release one with support for the pzem004t included.

I know it's possible because this sensor is supported.

I ask this because I'm not able to compile this myself because of lack of proper internet (just slow metered network available) paired with a lack of knowledge (which could be substituted if there wouldn't be the lack of internet :smirk_cat:)

Would be really helpful - not just for me but probably all people with limited access to internet!

Thank's a lot!

rradar, 2019-04-05:
Is there any chance that this one can go already into the 1.13.6 release?

Or maybe into @mcspr nightlies? I could get my Blithwolf SHP v2.3 running like this before it was officially released :smiley: 

I'm also willing to do the beta testing and report back!

Thank's a lot!

rradar, 2019-05-19:
@mcspr @xoseperez 

I tested the #1688 espurna-nodemcu-pzem004t binary successful with software serial.

Would it be possible to get this one (same same but different) also in the next 13.point release? I also can test the new binary.
I guess it will just work right away as nodemcu and wemos d1 mini are technically very close to each other ;)

mcspr, 2019-05-20:
> I guess it will just work right away as nodemcu and wemos d1 mini are
> technically very close to each other ;)

It should work right away! Only thing that we do care about from our
side if flash size, because build settings specify 4MB. AFAIK both
nodemcu and wemos d1mini are 4MB boards (or sometimes 16MB...)

So we don't need second 4MB variant, but another variant for 1MB flash size

rradar, 2019-06-08:
@mcspr you are right. Maybe the naming convention should be renewed. The images could be named `generic-1MB, generic-4MB` and a hint in the wiki that these work with generic esp8266 boards like the wemos d1 mini or the node mcu

 :+1:

davebuk, 2020-06-13:
I've just ordered a V3 PZEM004T and plan to use it with a wemos d1. I was going to use the TX/RX pins as the entry in `sensors.h` says software serial is not working. I guess that actually isn't the case and I can set software serial support and choose my own GPIO pins?