xoseperez/espurna #1586: Ticker.h error for  "rfb_ticker.attach_ms" - ITEAD_SONOFF_RFBRIDGE

kiralikbeyin, 2019-02-26:
Espurna Version [1.13.4] 2019-02-23
rc2.5.0

ITEAD_SONOFF_RFBRIDGE

**rfbridge.ino line 264**
        _rfb_ticker.attach_ms(RF_SEND_DELAY, _rfbSend);
to
        _rfb_ticker.attach_ms(RF_SEND_DELAY, **_rfbSendMe**);

**rfbridge.ino line 221**
void _rfbSend() {
to
void **_rfbSendMe**() {

reaper7, 2019-02-26:
complete error log for this issue:
```
C:\PROGRAMY\arduino_projekty\espurna_dev\code\espurna\rfbridge.ino: In function 'void _rfbSend(byte*, unsigned char)':

rfbridge:266:54: error: no matching function for call to 'Ticker::attach_ms(int, <unresolved overloaded function type>)'

         _rfb_ticker.attach_ms(RF_SEND_DELAY, _rfbSend);

                                                      ^

C:\PROGRAMY\arduino_projekty\espurna_dev\code\espurna\rfbridge.ino:266:54: note: candidates are:

In file included from C:\PROGRAMY\arduino_projekty\espurna_dev\code\espurna\mqtt.ino:16:0:

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\Ticker/Ticker.h:60:7: note: void Ticker::attach_ms(uint32_t, Ticker::callback_function_t)

  void attach_ms(uint32_t milliseconds, callback_function_t callback)

       ^

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\Ticker/Ticker.h:60:7: note:   no known conversion for argument 2 from '<unresolved overloaded function type>' to 'Ticker::callback_function_t {aka std::function<void()>}'

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\Ticker/Ticker.h:78:7: note: template<class TArg> void Ticker::attach_ms(uint32_t, void (*)(TArg), TArg)

  void attach_ms(uint32_t milliseconds, void (*callback)(TArg), TArg arg)

       ^

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\Ticker/Ticker.h:78:7: note:   template argument deduction/substitution failed:

C:\PROGRAMY\arduino_projekty\espurna_dev\code\espurna\rfbridge.ino:266:54: note:   candidate expects 1 argument, 2 provided

         _rfb_ticker.attach_ms(RF_SEND_DELAY, _rfbSend);

                                                      ^

C:\PROGRAMY\arduino_projekty\espurna_dev\code\espurna\rfbridge.ino:266:54: note:   candidate expects 1 argument, 0 provided

C:\PROGRAMY\arduino_projekty\espurna_dev\code\espurna\rfbridge.ino:266:54: note:   candidate expects 3 arguments, 2 provided

exit status 1
no matching function for call to 'Ticker::attach_ms(int, <unresolved overloaded function type>)'
```

Renaming  **_rfbSend**  to different name like **_rfbSendNoParams** solves problem.

kiralikbeyin, 2019-02-26:
Also light.ino have "attach_ms" function

line 694
```_light_transition_ticker.attach_ms(LIGHT_TRANSITION_STEP, _lightProviderUpdate);```

reaper7, 2019-02-26:
@kiralikbeyin - yes but function _lightProviderUpdate not have variants with parameters

mcspr, 2019-02-27:
*Just a note about this:*
If I remember right, new ticker API does some things via std::function which really does not like overloaded functions. Renaming works, yes.

PR #1693 (2019-05-08): RF/RFBRIDGE Update
fixes #1590 
fixes #1586 
maybe #1014 

- redefine RF_SUPPORT to be generic RF support. Use RFB_DIRECT=1 for RCSwitch implementation, RFB_DIRECT=0 for EFM8 controller.
- remove optional RF RAW support.
- rename RF_PIN -> RFB_RX_PIN, fix hardware and general headers
- aand small fix for Core 2.5.0 compilation.

I have only tried building itead-sonoff-rfbridge bin with RFB_DIRECT=0/1, to see if everything is in place. No live testing done yet.

cc @InduPrakash (https://github.com/xoseperez/espurna/issues/1014#issuecomment-482077321)
closed 2019-05-08 by mcspr