xoseperez/espurna #1589: Update to ArduinoJson 6

mcspr, 2019-02-27:
ref #924, #1179 

Since 6.9.0 ArduinoJson is no longer in beta.
https://github.com/bblanchon/ArduinoJson/releases/tag/v6.9.0

Niek, 2019-07-11:
Migration docs are available @ https://arduinojson.org/v6/doc/upgrade/

mcspr, 2019-07-24:
Important note from the migration guide:
```
// ArduinoJson 5
DynamicJsonBuffer jb;

// ArduinoJson 6
DynamicJsonDocument doc(1024);
```
> DynamicJsonDocument has a fixed capacity that you must specify to the constructor. Unlike the DynamicJsonBuffer, DynamicJsonDocument doesn’t automatically expand.

All dynamic memory buffers now need hard-coded max capacity value. Old DynamicJsonBuffer also had a fixed-size constructor option, but I think that was used here exactly once and since was removed from the code. Only 10 files use it:
https://github.com/xoseperez/espurna/search?l=C%2B%2B&q=DynamicJsonBuffer
HomeAssistant & WS are the main ones, which are known to crash specifically with json output. Web could also use some sane limits for the uploaded config.

Niek, 2019-07-31:
I don't know too much about the ArduinoJson code, but wouldn't it make sense to use ``StaticJsonDocument`` instead of ``DynamicJsonDocument``, if you're not going to expand beyond the initially allocated size? The way I read the docs, that should be faster (and possibly smaller).

mcspr, 2019-07-31:
IDK. Static one is a template, so we actually create a new class definition every time we use a different document size. Plus, it uses stack, and it might be problematic if there are some other heavy structs used.

*by default, esp8266 stack is 4k deep. if WPS is enabled, it is 8k (but -4k less heap, because of the Core tweaks).
still less than 50k of available heap*

mcspr, 2019-08-01:
There is also useful [`::memoryUsage()`](https://arduinojson.org/v6/api/jsondocument/memoryusage/) that calculates used size. To test things, one can put some random document size and call this after all values are set. Resulting value can then be hard-coded (ofc, taking into consideration const / non-const strings)

Size still grows. Current `nodemcu-lolin` build from v6 branch is 483472 bytes versus 477504 from the latest nightly.

mcspr, 2019-12-04:
Oh, I missed this release's changelog. While noticing the writer issue, its turns out there are readers too!

https://github.com/bblanchon/ArduinoJson/releases/tag/v6.13.0
> Added support for custom writer/reader classes (issue 1088)

Some weird things like ws incoming buffer could use non-linear memory a-la telnet outgoing buffer