xoseperez/espurna #1590: Compile error with SonOff rf bridge 433 RF_RAW_SUPPORT

Frans1960, 2019-02-27:
This is the Visual Studio compilation output with de errors with RF_RAW_SUPPORT = 1. How Can I solve the errors? Or is the latest source faulty?

error: 'isRFRaw' was not declared in this scope
if ((len > 0) && (isRFRaw || len != RF_MESSAGE_SIZE)) {
^
error: invalid conversion from 'const char*' to 'char*'
[-fpermissive]_rfbParseCode(payload);
^
error:   initializing argument 1 of 'void _rfbParseCode(char*)' [-fpermissive]
void _rfbParseCode(char * code) {
^

`Processing itead-sonoff-rfbridge (framework: arduino; platform: espressif8266@1.5.0; board: esp01_1m)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Converting espurna.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 54 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 5.13.4
|-- <AsyncMqttClient> 0.8.1 #ddbf4d1
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|-- <Brzo I2C> 1.3.2
|-- <DebounceEvent> 2.0.5 #fdc9c08
|-- <EEPROM_Rotate> 0.9.2 #00b9d6c
|   |-- <EEPROM> 1.0
|-- <Embedis> 1.2.0
|-- <Encoder>
|-- <EspSoftwareSerial> 3.4.1 #5378868
|-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |-- <ESP8266WiFi> 1.0
|-- <ESP Async WebServer> 1.1.0 #05306e4
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.4
|   |-- <Hash> 1.0
|-- <FauxmoESP> 3.1.0 #0614271
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <HLW8012> 1.1.0 #cc3baec
|-- <IRremoteESP8266> 2.2.0 #4d57a2b
|-- <JustWifi> 2.0.2 #1367a8b
|   |-- <ESP8266WiFi> 1.0
|-- <mDNSResolver> 0.1 #4cfcda1
|   |-- <ESP8266WiFi> 1.0
|-- <my92xx> 3.0.1 #cda124a
|-- <NoFUSS> 0.2.5 #1835206
|   |-- <ArduinoJson> 5.13.4
|   |-- <ESP8266httpUpdate> 1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266HTTPClient> 1.1
|   |   |   |-- <ESP8266WiFi> 1.0
|-- <NtpClientLib> 2.5.0 #0942ebc
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <Time> 1.5 #ecb2bb1
|   |-- <ESP8266WiFi> 1.0
|-- <OneWire> 2.3.4
|-- <PZEM004T> 1.1.2
|   |-- <EspSoftwareSerial> 3.4.1 #5378868
|-- <PubSubClient> 2.7
|-- <rc-switch> 2.6.2
[... 270 lines cut ...]
Environment bh-onofre                           [SKIP]
Environment bh-onofre-ota                       [SKIP]
Environment generic-ag-l4                       [SKIP]
Environment generic-ag-l4-ota                   [SKIP]
Environment lohas-e27-9w                        [SKIP]
Environment lohas-e27-9w-ota                    [SKIP]
Environment allterco-shelly1                    [SKIP]
Environment allterco-shelly1-ota                [SKIP]
Environment allterco-shelly2                    [SKIP]
Environment allterco-shelly2-ota                [SKIP]
Environment xiaomi-smart-desk-lamp              [SKIP]
Environment xiaomi-smart-desk-lamp-ota          [SKIP]
Environment phyx-esp12-rgb                      [SKIP]
Environment phyx-esp12-rgb-ota                  [SKIP]
Environment bestek-mrj1011                      [SKIP]
Environment bestek-mrj1011-ota                  [SKIP]
Environment gblife-rgbw-socket                  [SKIP]
Environment gblife-rgbw-socket-ota              [SKIP]
Environment smartlife-mini-smart-socket         [SKIP]
Environment smartlife-mini-smart-socket-ota     [SKIP]
================================================= [ERROR] Took 8.19 seconds =================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
`

olili, 2019-03-02:
same problem. Any solution available?

Frans1960, 2019-03-02:
Yes, take the quality release!

PR #1693 (2019-05-08): RF/RFBRIDGE Update
fixes #1590 
fixes #1586 
maybe #1014 

- redefine RF_SUPPORT to be generic RF support. Use RFB_DIRECT=1 for RCSwitch implementation, RFB_DIRECT=0 for EFM8 controller.
- remove optional RF RAW support.
- rename RF_PIN -> RFB_RX_PIN, fix hardware and general headers
- aand small fix for Core 2.5.0 compilation.

I have only tried building itead-sonoff-rfbridge bin with RFB_DIRECT=0/1, to see if everything is in place. No live testing done yet.

cc @InduPrakash (https://github.com/xoseperez/espurna/issues/1014#issuecomment-482077321)
closed 2019-05-08 by mcspr