xoseperez/espurna #1594: Sonoff T1 Touch Lockup

Fusseldieb, 2019-02-28:
**Bug description**
If I press the touch buttons multiple times in a row, it will lock up after 3-4 times and only respond after 1-2 seconds again. This happens with the touch buttons, in the Web UI and inside HassIO (when configured over MQTT).

**Steps to reproduce**
Flash it, let it boot up and then press the touch buttons repeatedly until it locks up for 1-2 seconds, normally this happens after 3-4 presses.

**Expected behavior**
That no matter how many times I press it, it doesn't lock up after 3-4 times.

**Device information**
* *ESPurna version* 1.13.4
* *Flash mode* Serial FTDI (GPIO0+GND -> Erase flash -> GPIO0+GND -> Espurna T1 bin)
* *Device brand, model and version* SONOFF T1 1.0 US (ESP8285)

mcspr, 2019-02-28:
You sure it locks up and not resetting? If so, i think it is intended behaviour:
https://github.com/xoseperez/espurna/blob/5c0abca719f4151c294c7277da4baf9d8629a9c4/code/espurna/config/general.h#L339-L347
i.e. it will allow 5 changes in 3 second window. You can check the telnet / serial log for lines like this:
```
[427429] [RELAY] #1 scheduled ON in 657 ms
```

And you might want to update to 1.13.5 if it resets instead.

Fusseldieb, 2019-02-28:
> You sure it locks up and not resetting? If so, i think it is intended behaviour:
Yes, it's locking up, not resetting, since the WiFi doesn't vanish and the WebUI reponds immediately after 1-2 seconds again.
> i.e. it will allow 5 changes in 3 second window.
I didn't knew that ESPURNA had a flood protection... That could be it. 

A option to disable it via the WebUI would be nice.

Thanks

mcspr, 2019-03-01:
Sure, web will support it. I would not put a definitive timeline on this, but soon (tm)

And may I ask - what is the use case?

Fusseldieb, 2019-03-01:
> Sure, web will support it. I would not put a definitive timeline on this, but soon (tm)

That's nice

> And may I ask - what is the use case?

It just feels unresponsive somehow, I can't really explain it better.

xoseperez, 2019-03-21:
Flood protection was initially implemented to protect mechanical relays. But since the code is in the core of the `relay.ino` module it actually affect all type of switches, including light bulbs or triacs. Having an option to disable or to define a default value per-device will be great.

closed 2019-11-09 by mcspr