xoseperez/espurna #1597: Sonoff RF Bridge cant change "Boot Mode" of switch

kiralikbeyin, 2019-03-01:
Espurna 1.13.5
rc2.5.0 and rc2.4.2

When i  want to change "Always OFF" to "Same as before" it doesnt save and restarts... 
So my switches turns off everytime SRFB restarts.

reaper7, 2019-03-01:
looks like common problem with "save".
My S20 and RF Bridge do not save settings.
Only Sonoff T1 save settings but I do not know how I forced him to do it...
It seems to me that I had to enable the following options:
TELNET_SUPPORT & DEBUG_TELNET_SUPPORT & DEBUG_WEB_SUPPORT
from this moment pressing the SAVE key on the web interaface began to save the settings.

Espurna current git / LWIP 1.4

mcspr, 2019-03-01:
Can't reproduce with rfbridge built with 2.4.2, with or without DEBUG_SUPPORT flags active.
Does using lwip2 make any difference?
Does setting `LOOP_DELAY_TIME` to `10` change anything? (or without rebuilding - `loopDelay` setting via terminal / settings upload)

kiralikbeyin, 2019-03-02:
@reaper7 yes, save problem. TELNET_SUPPORT & DEBUG_TELNET_SUPPORT & DEBUG_WEB_SUPPORT didnt work Espurna current git / LWIP 1.4 AND Espurna current git / V2 Low mem didnt work.

@mcspr loopDelay 10, 20, 55, 200 didnt work

I have a IR RECEIVE problem maybe these problems are related. https://github.com/xoseperez/espurna/issues/1596

I will fallback to 1.13.3 v

kiralikbeyin, 2019-03-02:
I uploaded espurna-1.13.5-itead-sonoff-rfbridge.bin with esptool.py and it saves no reboot.

I think main problem is we are downloading external libraries like https://github.com/markszabo/IRremoteESP8266 and other 10-15 libraries and they are crushing something. (i downloaded all latest external libs and i cant save which is the most basic thing)

**Please add external libaries with exact relase in source.zip of nice Espurna.**
Also please give info which "rc version" and "lwIP Variant" is used for these bin files.
@xoseperez Thanks alot!

Is there any way to enable IR receive-send from terminal?

reaper7, 2019-03-02:
my observations:

---

custom config:
```
  #define WIFI_FALLBACK_APMODE        0                                         // Fallback to AP mode if no STA connection
  #define USE_PASSWORD                0

  #define TELNET_SUPPORT              0                                         // Enable telnet support by default (3.34Kb)
  #define TELNET_STA                  0                                         // By default, disallow connections via STA interface
  #define TELNET_AUTHENTICATION       0                                         // Request password to start telnet session by default
  
  #define TERMINAL_SUPPORT            0                                         // Enable terminal commands (0.97Kb)

  #define WEB_SUPPORT                 1                                         // Enable web support (http, api, 121.65Kb)
  #define WEB_EMBEDDED                1                                         // Build the firmware with the web interface embedded in
  #define WEB_FORCE_PASS_CHANGE       0

  #define DEBUG_SERIAL_SUPPORT        0                                         // Enable serial debug log
  #define DEBUG_UDP_SUPPORT           0                                         // Enable UDP debug log
  #define DEBUG_TELNET_SUPPORT        0                                         // Enable telnet debug log (will only work if TELNET_SUPPORT is also 1)
  #define DEBUG_WEB_SUPPORT           0                                         // Enable web debug log (will only work if WEB_SUPPORT is also 1)

  #define SCHEDULER_SUPPORT           0                                         // Enable scheduler (1.77Kb)

  #define NTP_SUPPORT                 1                                         // Build with NTP support by default (6.78Kb)
  #define NTP_SERVER                  "myhouse.lan"

  #define MQTT_SUPPORT                0                                         // MQTT support (22.38Kb async, 12.48Kb sync)
  #define MQTT_ENABLED                1                                         // Do not enable MQTT connection by default
  #define MQTT_AUTOCONNECT            1                                         // If enabled and MDNS_SERVER_SUPPORT=1 will perform an autodiscover and
  #define MQTT_SERVER                 "rpi"                                     // Default MQTT broker address
  #define MQTT_USER                   "user"                                    // Default MQTT broker usename
  #define MQTT_PASS                   "pass"                                    // Default MQTT broker password
  #define MQTT_RETAIN                 true                                      // MQTT retain flag
  #define MQTT_QOS                    0                                         // MQTT QoS value for all messages
  #define MQTT_USE_JSON               0                                         // Group messages in a JSON body

  #define HOMEASSISTANT_SUPPORT       0                                         // Build with home assistant support (if MQTT, 1.64Kb)
  #define HOMEASSISTANT_ENABLED       1                                         // Integration not enabled by default
  #define HOMEASSISTANT_PREFIX        MY_HOME                                   // Default MQTT prefix

  #define DOMOTICZ_SUPPORT            0                                         // Build with domoticz (if MQTT) support (1.72Kb)
  #define THINGSPEAK_SUPPORT          0                                         // Enable Thingspeak support by default (2.56Kb)
  #define ALEXA_SUPPORT               0                                         // Enable Alexa support by default (10.84Kb)
```

with the same settings, I enable and disable mqtt support only (MQTT_SUPPORT):

---

with MQTT_SUPPORT **disabled** all works, I can SAVE, when SAVE button pressed wireshark show:

[image]

main window:
[image]

---

with MQTT_SUPPORT **enabled** I **can't** SAVE, when SAVE button pressed wireshark show:

[image]

main window:
[image]

---

arduino libs used:
```
Użycie biblioteki ArduinoJson-5.11.2 w wersji 5.11.2 z folderu: C:\PROGRAMY\arduino_projekty\libraries\ArduinoJson-5.11.2 
Użycie biblioteki eeprom_rotate w wersji 0.9.2 z folderu: C:\PROGRAMY\arduino_projekty\libraries\eeprom_rotate 
Użycie biblioteki EEPROM w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\EEPROM 
Użycie biblioteki ESPAsyncTCP w wersji 1.2.0 z folderu: C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncTCP 
Użycie biblioteki embedis w wersji 1.2.0 z folderu: C:\PROGRAMY\arduino_projekty\libraries\embedis 
Użycie biblioteki ESPAsyncWebServer w wersji 1.2.0 z folderu: C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncWebServer 
Użycie biblioteki ESP8266WiFi w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
Użycie biblioteki justwifi w wersji 2.0.2 z folderu: C:\PROGRAMY\arduino_projekty\libraries\justwifi 
Użycie biblioteki debounceevent w wersji 2.0.5 z folderu: C:\PROGRAMY\arduino_projekty\libraries\debounceevent 
Użycie biblioteki ESP8266mDNS w wersji 1.2 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS 
Użycie biblioteki Time w wersji 1.5 z folderu: C:\PROGRAMY\arduino_projekty\libraries\Time 
Użycie biblioteki NtpClient w wersji 2.5.0 z folderu: C:\PROGRAMY\arduino_projekty\libraries\NtpClient 
Użycie biblioteki Ticker w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\Ticker 
Użycie biblioteki ArduinoOTA w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\ArduinoOTA 
Użycie biblioteki Hash w wersji 1.0 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\Hash 
Użycie biblioteki DNSServer w wersji 1.1.1 z folderu: C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\libraries\DNSServer 
```

arduino settings:
[image]

kiralikbeyin, 2019-03-02:
@reaper7 thanks for showing all info. Did you try https://github.com/xoseperez/espurna/releases/download/1.13.5/espurna-1.13.5-itead-sonoff-rfbridge.bin

I enabled MQTT everything is fine, i can save.

reaper7, 2019-03-02:
@kiralikbeyin - it's better for me to find the cause, because I use custom config with predefined settings for each of my devices (ip addresses, relays boot modes, way the diodes works, topics for mqtt etc...)

kiralikbeyin, 2019-03-02:
@reaper7 @mcspr 
Found solution - cut USB to GPIO 4-5 connection 

Also i can receive IR codes :)

more info :https://github.com/arendst/Sonoff-Tasmota/issues/5379

closed 2019-03-02 by kiralikbeyin

mcspr, 2019-03-02:
@kiralikbeyin binaries are built by platformio. libraries are pinned, like this: https://github.com/xoseperez/espurna/blob/e62d0df9be411ae1e30a8c8d4e8ae8d70f0d429a/code/platformio.ini#L89
and it is using 2.3.0 core, single option of lwip1.4. glad that you solved your problem with ir lib. so does it mean that the latest version works ok and that was some kind of power problem?

But, @reaper7 mentioned the same problem with different devices.
since device is seemingly crashing, here's the custom build based on 1.13.5 with `/crash` http endpoint to get the stacktrace for https://github.com/me-no-dev/EspExceptionDecoder (just make sure to keep .bin / .elf from the build directory before restarting IDE)
[.bin with custom config above, minus personalized server settings](https://github.com/xoseperez/espurna/files/2922302/itead-sonoff-rfbridge-check-1.13.5.zip)

kiralikbeyin, 2019-03-02:
@mcspr Thanks for information i never tried rc2.3 or IR REMOTE LIB v2.2.0 (20170922) but i will check this file everytime in the future.

Strange thing about this lib https://github.com/1technophile/OpenMQTTGateway/issues/367 works fine with USB power (without cutting) and also FTDI connection.

When i cut USB -GPIO4-GPIO5 connection Espurna and Tasmato libs also work (IR receive)

My strange idea for this strange problem is esp chip needs more power if it loops more code like MQTT SUPPORT :) (Note that OpenMQTTGateway is smaller than Espurna / Tasmota )

@reaper7 cut the USB-GPIO4-5 connection and you will save your config

mcspr, 2019-03-02:
Yeah, that is strange.
OpenMQTTGateway uses the same IR library, but version 2.5.5:
https://github.com/1technophile/OpenMQTTGateway/tree/development/lib/IRremoteESP8266
And there's the same flavour of MQTT PubSubClient in ESPurna, by adding `#define MQTT_USE_ASYNC 0` and setting `MQTT_MAX_PACKET_SIZE` to `1024` in PubSubClient sources.

reaper7, 2019-03-02:
@mcspr - looks like you've solved my problem!
#define MQTT_USE_ASYNC 0
this is a solution for me 👍 

now all my sonoff devices saves changes :)

[image]

mcspr, 2019-03-02:
Ok, assuming MQTT client is at fault then. Probably related to the https://github.com/xoseperez/espurna/issues/1115
Minimal test with web debugging tools console:
```js
> websock.send('{"config": {"mqttEnabled": 1, "mqttServer": "...server address..."}}')
```
I don't see any crashes though.

reaper7, 2019-03-03:
@mcspr - mosquitto 1.4.10-3 (with user/pass!!!) on rpi

mcspr, 2019-03-04:
Well, turns out it is not about the server. https://github.com/xoseperez/espurna/issues/1115#issuecomment-413833825 already described the solution for the mqtt client codebase. pr for async-mqtt-client is still hanging there.

The culprit is lwip's tcp_output, which does not check if connection is closed. While inside websocket callback, mqttReset() does immediately disconnect leaving connection object NULL and causing the crash. Ofcourse, doing the exact same thing inside loop() + yield() right after is a possible workaround (while making things somewhat slower...)

Also note, that this only affects git version. Release version of async-mqtt-client is still ok, and is being used by the platformio builds.