xoseperez/espurna #1606: Cannot switch on relay on SonOFF Basic after upgrading to 1.13.6-dev

karmacoma92, 2019-03-04:
**Bug description**
Cannot switch on relay on SonOFF Basic after upgrading to 1.13.6-dev

**Steps to reproduce**
just upgraded from 1.13.4. Trying to check if it was because of Issue [#409](https://github.com/xoseperez/espurna/issues/409), I have found that it works properly after switching MQTT off 

and this is when it doesn't work

[303722] [WEBSOCKET] Requested action: relay
[303724] [RELAY] #0 scheduled ON in 0 ms
[303727] [RELAY] #0 set to ON
[303731] [MQTT] Sending domoticz/in => {"idx": 15, "nvalue": 1, "svalue": ""} (PID 1)
[303735] [MQTT] Sending ESPURNA-A70ED8/relay/0 => 1 (PID 1)
[303874] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "0001405F",
   "idx" : 15,
   "name" : "SonOFF-2-Switch",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

[303894] [DOMOTICZ] Received value 1 for IDX 15
[303896] [RELAY] #0 scheduled OFF in 0 ms
[303898] [RELAY] #0 set to OFF
[303903] [MQTT] Sending ESPURNA-A70ED8/relay/0 => 0 (PID 1)
[304909] [RELAY] Setting relay mask: 0

**Expected behavior**

this is a log from the debug when it works properly

[021253] [WEBSOCKET] Requested action: relay
[021256] [RELAY] #0 scheduled ON in 0 ms
[021259] [RELAY] #0 set to ON
[022263] [RELAY] Setting relay mask: 1
[035527] [WEBSOCKET] Requested action: relay
[035529] [RELAY] #0 scheduled OFF in 0 ms
[035532] [RELAY] #0 set to OFF
[036536] [RELAY] Setting relay mask: 0
[060001] [MAIN] System OK

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
---8<-------

[399509] [MAIN] ESPURNA 1.13.6.PA3-dev (190302)
[399510] [MAIN] xose.perez@gmail.com
[399511] [MAIN] http://tinkerman.cat

[399515] [MAIN] CPU chip ID: 0xA70ED8
[399517] [MAIN] CPU frequency: 80 MHz
[399518] [MAIN] SDK version: 2.2.1(cfd48f3)
[399521] [MAIN] Core version: 2.4.2
[399524] [MAIN] Core revision: 3140015267
[399530] 
[399532] [MAIN] Flash chip ID: 0x14405E
[399535] [MAIN] Flash speed: 40000000 Hz
[399539] [MAIN] Flash mode: DOUT
[399541] 
[399543] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[399548] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[399554] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[399561] [MAIN] Firmware size       :   523088 bytes /  128 sectors (   1 to  128)
[399568] [MAIN] Max OTA size        :   499712 bytes /  122 sectors ( 129 to  250)
[399579] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[399583] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[399590] 
[399591] [MAIN] EEPROM sectors: 251, 250
[399594] [MAIN] EEPROM current: 251
[399597] 
[399598] [MAIN] EEPROM:  4096 bytes initially |  1477 bytes used (36%) |  2619 bytes free (63%)
[399607] [MAIN] Heap  : 39184 bytes initially | 22480 bytes used (57%) | 16704 bytes free (42%)
[399615] [MAIN] Stack :  4096 bytes initially |  1388 bytes used (33%) |  2708 bytes free (66%)
[399625] 
[399627] [MAIN] Boot version: 31
[399628] [MAIN] Boot mode: 1
[399631] [MAIN] Last reset reason: Reboot from web interface
[399637] 
[399639] [MAIN] Board: ITEAD_SONOFF_BASIC
[399640] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB 
[399658] [MAIN] Sensors: DHTXX 
[399660] [MAIN] WebUI image: SENSOR
[399661] 
[399664] [MAIN] Firmware MD5: 55422ca17dc500db797eb80ace858c07
[399668] [MAIN] Power: 3303 mV
[399672] [MAIN] Power saving delay value: 1 ms
[399675] 

---8<-------

[399676] +OK

**Tools used**
Ubuntu 18.04.2
Firefox 65.0.1
PlatformIO Home 2.0.0, Core 3.6.4

**Additional context**

mcspr, 2019-03-05:
@karmacoma92 can you try changing https://github.com/xoseperez/espurna/blob/f74e0b06612b1a13df4b324d69a17db43001867f/code/espurna/domoticz.ino#L142
from `>` to `>=`? pretty sure it is caused by that.

karmacoma92, 2019-03-05:
hi Max
it works, both in a SonOFF Basic and a SonOFF TH
thanks
Oscar

commit 07ba3ff2 (2019-03-07): domoticz: fix basic switch nvalue

fixes #1606

When ON domoticz sends:
value >= 1 for lights
value == 1 for basic switches
closed 2019-03-07 by mcspr