xoseperez/espurna #1613: Compile error - ArduinoJson

icevoodoo, 2019-03-07:
I get those errors bellow when I try to compile Espurna for Sonoff-TH. I use Visual Studio Code. Untill 2-3 day ago everything was OK. Now the latest arduinojson has broke my compilation process.

Please advice, and 10x for Your time.

```
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1169:73: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject& {aka ArduinoJson691_000::ObjectRef&}' from an rvalue of type 'ArduinoJson691_000::ObjectRef'
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1181:59: error: no matching function for call to 'ArduinoJson691_000::ObjectRef::set(char*&, int)'Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\ESP8266WiFiScan.cpp.o

D:/GitHub/xoseperez/code/espurna/mqtt.ino:1181:59: note: candidate is:
In file included from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.hpp:16:0,
from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.h:9,
from espurna\config/prototypes.h:2,
from espurna\config/all.h:35,
Compiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\WiFiClient.cpp.o
from D:/GitHub/xoseperez/code/espurna/espurna.ino:43:
.piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Object/ObjectRef.hpp:166:21: note: bool ArduinoJson691_000::ObjectRef::set(ArduinoJson691_000::ObjectConstRef)
FORCE_INLINE bool set(ObjectConstRef src) {
^
.piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Object/ObjectRef.hpp:166:21: note:   candidate expects 1 argument, 2 provided
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1185:54: error: no matching function for call to 'ArduinoJson691_000::ObjectRef::set(char*&, double&)'
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1185:54: note: candidate is:
In file included from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.hpp:16:0,
from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.h:9,
from espurna\config/prototypes.h:2,
from espurna\config/all.h:35,
from D:/GitHub/xoseperez/code/espurna/espurna.ino:43:
.piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Object/ObjectRef.hpp:166:21: note: bool ArduinoJson691_000::ObjectRef::set(ArduinoJson691_000::ObjectConstRef)
FORCE_INLINE bool set(ObjectConstRef src) {
^
.piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Object/ObjectRef.hpp:166:21: note:   candidate expects 1 argument, 2 provided
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1191:60: error: no matching function for call to 'ArduinoJson691_000::ObjectRef::set(char*&, char*&)'
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1191:60: note: candidate is:
In file included from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.hpp:16:0,
from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.h:9,
from espurna\config/prototypes.h:2,
from espurna\config/all.h:35,
from D:/GitHub/xoseperez/code/espurna/espurna.ino:43:
.piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Object/ObjectRef.hpp:166:21: note: bool ArduinoJson691_000::ObjectRef::set(ArduinoJson691_000::ObjectConstRef)
FORCE_INLINE bool set(ObjectConstRef src) {
^
.piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Object/ObjectRef.hpp:166:21: note:   candidate expects 1 argument, 2 provided
D:/GitHub/xoseperez/code/espurna/mqtt.ino: In function 'void mqttFlush()':
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1223:23: error: 'jsonBuffer' was not declared in this scope
D:/GitHub/xoseperez/code/espurna/mqtt.ino:1269:10: error: 'ArduinoJson::JsonObject' has no member named 'printTo'
D:/GitHub/xoseperez/code/espurna/relay.ino: In function 'void _relayWebSocketUpdate(ArduinoJson::JsonObject&)':
D:/GitHub/xoseperez/code/espurna/relay.ino:1189:60: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino: In function 'void _relayWebSocketSendRelays()':
D:/GitHub/xoseperez/code/espurna/relay.ino:1261:23: error: 'jsonBuffer' was not declared in this scope
D:/GitHub/xoseperez/code/espurna/relay.ino:1265:63: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject& {aka ArduinoJson691_000::ObjectRef&}' from an rvalue of type 'ArduinoJson691_000::ObjectRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1275:54: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1277:54: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1279:56: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1281:54: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1283:56: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1285:66: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1291:60: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1293:70: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/relay.ino:1295:70: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
D:/GitHub/xoseperez/code/espurna/scheduler.ino: In function 'void _schWebSocketOnSend(ArduinoJson::JsonObject&)':
DC:/GitHub/xoseperez/code/espurna/scheduler.ino:57:64: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject& {aka ArduinoJson691_000::ObjectRef&}' from an rvalue of type 'ArduinoJson691_000::ObjectRef'ompiling .pioenvs\itead-sonoff-th\lib537\ESP8266WiFi\WiFiClientSecure.cpp.o

schedules["start"] = 0;
^
D:/GitHub/xoseperez/code/espurna/scheduler.ino:63:66: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
// -----------------------------------------------------------------------------
^
D:/GitHub/xoseperez/code/espurna/scheduler.ino:65:65: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
void _schConfigure() {
^
D:/GitHub/xoseperez/code/espurna/scheduler.ino:67:64: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray& {aka ArduinoJson691_000::ArrayRef&}' from an rvalue of type 'ArduinoJson691_000::ArrayRef'
[... 651 lines cut ...]
Environment bh-onofre                           [SKIP]
Environment bh-onofre-ota                       [SKIP]
Environment generic-ag-l4                       [SKIP]
Environment generic-ag-l4-ota                   [SKIP]
Environment lohas-e27-9w                        [SKIP]
Environment lohas-e27-9w-ota                    [SKIP]
Environment allterco-shelly1                    [SKIP]
Environment allterco-shelly1-ota                [SKIP]
Environment allterco-shelly2                    [SKIP]
Environment allterco-shelly2-ota                [SKIP]
Environment xiaomi-smart-desk-lamp              [SKIP]
Environment xiaomi-smart-desk-lamp-ota          [SKIP]
Environment phyx-esp12-rgb                      [SKIP]
Environment phyx-esp12-rgb-ota                  [SKIP]
Environment bestek-mrj1011                      [SKIP]
Environment bestek-mrj1011-ota                  [SKIP]
Environment gblife-rgbw-socket                  [SKIP]
Environment gblife-rgbw-socket-ota              [SKIP]
Environment smartlife-mini-smart-socket         [SKIP]
Environment smartlife-mini-smart-socket-ota     [SKIP]
Environment hama-wifi-steckdose-00176533        [SKIP]
Environment hama-wifi-steckdose-00176533-ota    [SKIP]
================================================== [ERROR] Took 28.74 seconds ==================================================
The terminal process terminated with exit code: 1
```

karmacoma92, 2019-03-07:
It looks like ArduinoJson library has moved to version 6 and it is not supported by espurna yet, I had the same issue today and fixed it by replacing  

lib_deps =
    ArduinoJson

with 

lib_deps =
    https://github.com/bblanchon/ArduinoJson#v5.13.5

in platformio.ini

do not know if it is the preferred or more canonical way to sort this out but it works

I hope it helps
Oscar

icevoodoo, 2019-03-08:
> It looks like ArduinoJson library has moved to version 6 and it is not supported by espurna yet, I had the same issue today and fixed it by replacing
>
> lib_deps =
> ArduinoJson
>
> with
>
> lib_deps = [...]

Thanks you are right , it works. 
Hope this will be resolved soon to work with the new ArduinoJson 6.

closed 2019-03-08 by icevoodoo

PetrJakes, 2019-03-23:
Worked to  me as well, thank you