xoseperez/espurna #1614: Gosund SP1 and 1.13.5

jbaumann, 2019-03-07:
I used tuya-convert to install Espurna 1.13.5 on two Gosund power plugs (Tuya power plugs). I used the firmware 
https://github.com/xoseperez/espurna/releases/download/1.13.5/espurna-1.13.5-gosund-sp1-v23.bin

This worked without problems. After a hard reset (pulling it from the socket) I was able to log into the AP, configure a new password, restart the power plug, configure another name and theWLAN, recalibrate the power consumption and everything worked (with both plugs). During the configuration I also at some time rebooted the power plugs.

After removing the plugs I wanted to use them one day later.

The devices connected to the WLAN using the configured name. When I tried to log into them, they asked for the credentials, then showed an empty page with a bomb icon. The switch works and turns the relay on and off (LED changes accordingly).

I reset one of the devices to factory values and it opens the AP, as it should. When providing the credentials though, it again shows the empty page with the bomb icon.

My first hypothesis: The image is too large and is partly overwritten when recalibrating the power measurements. Can I provide additional data to verify this?

My real question: Can I recover the devices without opening them? Is it possible to start an OTA in this situation?

Cheers, Joachim

jbaumann, 2019-03-09:
Ok, after quite some experiments including factory resets and uploading new versions of the same firmware using the HTTP-API I discovered that in fact everything works fine (the bomb image seems to be the normal espurna icon, and I simply ignored it before).

I switched computers between the tests and my old Macbook Air with High Sierra shows the Espurna page without problems, the newer one with Mojave does not (independent of modus i.e., with inkognito modus the same happens). This is something that is new (I just tested this with espurna 1.12.6 and with 1.13.3 and everything works on both computers). In some cases the browser on Mojave starts to download a file named download, suggesting that maybe the content type is not correct

I would label this a bug, and if I can provide additional data, I'm happy to do experiments.

Cheers, Joachim

jbaumann, 2019-03-09:
Next datapoint: I tried accessing the web Fronten with chrome on a Win 10 notebook, same behaviour as on the Mojave browser.

mcspr, 2019-03-09:
Hey

What does the Network tab show for `/ws` and `/` in the browser's web-dev tools? Both Safari and Chrome.
Does websocket frames list look incomplete like in #1610 ?

jbaumann, 2019-03-10:
Comparing the results of the High-Sierra- and the Mojave-Macbook.

"/"
Chrome HighSierra:
Status Code 200 OK
normal response headers (Content-Encoding, length, type etc are set)
Timing 9,65ms

Chrome Mojave
Status Code 304 Not Modified
Response Headers Connection close, content-length 0
Timing: 30.68ms

Safari Mojave
No request headers
no response headers
no timing data

"/ws"
Chrome HighSierra
Status code 101 Switching Protocols
Response headers look normal (Connection: Upgrade, Content-Length 0, Sec-Websocket accept..., Upgrade: websocket)
a number of messages ({"weblog":{"message"....}, {"magnitude":...})
Timing 2.3min

Chrome Mojave
Status code 101 Switching Protocols
Response headers look normal to me (Connection: Upgrade, Content-Length 0, Sec-Websocket accept..., Upgrade: websocket)
One message ({"relayConfig":...})
Timing "Caution: request not finished yet"

Safari Mojave
Status code 101 Switching Protocols
Response headers look normal to me (Connection: Upgrade, Content-Length 0, Sec-Websocket accept..., Upgrade: websocket)
Preview: Websocket connection initiated
Timing 0.3ms begin to end
sent and received bytes are 0

What else can I provide?

Cheers, Joachim

mcspr, 2019-03-10:
No, I think this pretty much explains it all. Since Chrome does not work too, I would assume it is exactly the same issue as #1610 and tcp fragmentation is the main culprit. What I don't understand is why Windows machine does the same. What build of Win10 do you use?

Just to confirm, you can access the device via `nc <ip> 23` you can track the memory usage via `heap` command. Tests that I made earlier showed that heap is steadily decreasing ~ -2.8Kb after every page reload and does not come back. When using browsers on Mojave I would expect that it happens too.

note: Telnet LAN connections are rejected by default, so that only works in softAP mode. Double-click on power button should bring it up. Once connected, we can avoid that restriction for the time being by `set telnetSTA 1` & `reload`

jbaumann, 2019-03-10:
The Windows system has Update 1809 installed. I just tested again, the same with Chrome, but with Internet Explorer it works...

I watched the heap (I turned telnet on using the WEB UI, shouldn't make a difference?), but couldn't see the same. I started with 17720 bytes used (50%), went to 14696 after the first failed access, then to 24360, and back to 14696...

mcspr, 2019-03-10:
Ok, returning to WS data, without any packet analysis via wireshark yet. Maybe some extension is messing things up?

For example, opening web in chrome on win10. `/ws` receives following frames: time, bytes received, data: 
```
:27.470  100 {"weblog":{"message":"[WEBSOCKET] #2 connected, ip: ..., url: /ws\n","prefix":"[894769] "}}
:27.471  170 {"relayConfig":{"size":1,...
:27.488 2860 {"maxNetworks":5,"wifiScan":false,"wifi":[..., "webMode": 0,....
:27.544   86 {"heap":11760,"uptime":18894,"rssi":-64,"loadaverage":3,"vcc":3232,"now":1552252407}
:30:400  428 {"magnitudes":{"...
```
webMode=0 kv in the third frame un-display:none s the page so that is a requirement for web to show up.
Chrome Mojave "One message ({"relayConfig":...})", so no web.
Safari Mojave "sent and received bytes are 0" is even stranger though. Maybe we are missing some new setting there?

jbaumann, 2019-03-14:
If I remember correctly I ran the tests in incognito tabs, and I think I have most plug-ins disabled there. Since I never use Safari it is not configured with any plug-ins in any way, I think. Do you need me to do some other tests?

mcspr, 2019-03-16:
If you can, please do. I'd focus on not-working macOS. 

The main questions are:
- is this network-stack related? does Firefox work?
maybe it is worth checking Wireshark output? `ip.addr == ... and tcp.port == 80` for filtering
- does the browser download `/` after clearing cache? (r-click on the request in the webdev tools)
- whether it is MTU issue (changed between OS versions or manually, some VPN software?)
https://github.com/xoseperez/espurna/issues/1610#issuecomment-473339454 is one way I was able to reproduce bug with Windows. Something like `curl -v -o check.html --digest -u admin:password http://.../` would not work too and will just hang forever. `curl -v -o check.html --digest -u admin:password http://.../auth` should still work.

jbaumann, 2019-03-17:
This might be something. I had an MTU of 1453 on the newer OSX computer. When I change that to 1500, everything works fine...
If this information doesn't solve the problem then I will do the other tests during the week.

mcspr, 2019-03-18:
Thanks!
So does this mean it was set automatically (DHCP setting, MTU discovery thing)? Does the router have the same MTU then? I'd call it a duplicate of #1620, to avoid having two issues.

Two external solutions atm are changing MTU size on the client, if that does not cause other issues, or building firmware using the newest Arduino Core + lwip2 option (which properly fragments packets in all cases, ref: tcpdump on the router)
Since this is easily reproducible, I'll try to change some code to make it work with the current Core version.

jbaumann, 2019-03-18:
I set it manually some time ago to get rid of spurious problems with a device and never changed it back. The router has 1500 as MTU. The lwip bug to me sounds like an exact description of what happens, and I would see this as a duplicate of #1610 .

mcspr, 2019-03-18:
Closing.

Small patch from lwip2 applied to the lwip1 worked for me on LAN when MTU is changed. Just to double check, will try with OpenVPN and add the patch to the code here so releases can be built with it. Second issue got the explanation about how to apply it to the Arduino Core lwip manually.

closed 2019-03-18 by mcspr