xoseperez/espurna #1615: Power meter with new chip

dpeddi, 2019-03-07:
Recently I got a couple of power meter PMB-01 1.6 externally identical to yours but it have onboard another chip:

BL6523GX

[001450378.pdf](https://github.com/xoseperez/espurna/files/2940020/001450378.pdf)

I've found the data sheet attached. It have a counter output and a tx/tx at 4800 baud to access registers.
It seems not to bad but still unsupported.

jvstch, 2022-02-19:
I recently added bl6523gx support to tasmota mainline and is currently coding the support in espurna, appreciate if someone could help with testing.

Tasmota support already in development branch.

some more details here: [https://youtu.be/s3rMb4opuDg](url)
[tasmota.bin.gz](https://github.com/xoseperez/espurna/files/8101321/tasmota.bin.gz)

Tasmota firmware with bl6523gx support attached

jvstch, 2022-02-23:
In progress at https://github.com/jvstch/espurna/tree/feature/Add-BL6523GX. Will post once a testable version is ready.