xoseperez/espurna #1621: Xiaomi Desk Lamp - Encoder Read Correctly, But Doesn't turn on

thehellodonkey, 2019-03-09:
Thanks for the great firmware, look forward to your thoughts on the issue below.

**Bug description**
Light doesn't turn on regardless of encoder inputs or switch/brightness/color settings in web interface.

**Steps to reproduce**
Load firmware on Xiaomi Smart Desk Lamp, attempt to turn on.

**Expected behavior**
Light on/off status reflects switch position in web interface

**Screenshots**
This shows web interface responding as expected to all encoder actions, but light does not turn on.
[image]

**Device information**
[753816] [MAIN] ESPURNA 1.13.5 (83195fd)
[753817] [MAIN] xose.perez@gmail.com
[753818] [MAIN] http://tinkerman.cat

[753819] [MAIN] CPU chip ID: 0xA30B3B60
[753821] [MAIN] CPU frequency: 80 MHz
[753825] [MAIN] SDK version: 1.5.3(aec24ac9)
[753829] [MAIN] Core version: 2.3.0
[753832] [MAIN] Core revision: 159542381
[753835] 
[753836] [MAIN] Flash chip ID: 0x1540C8
[753841] [MAIN] Flash speed: 40000000 Hz
[753843] [MAIN] Flash mode: DOUT
[753846] 
[753848] [MAIN] Flash size (CHIP)   :  2097152 bytes /  512 sectors (   0 to  511)
[753860] [MAIN] Flash size (SDK)    :  2097152 bytes /  512 sectors (   0 to  511)
[753862] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[753871] [MAIN] Firmware size       :   501840 bytes /  123 sectors (   1 to  123)
[753879] [MAIN] Max OTA size        :   520192 bytes /  127 sectors ( 124 to  250)
[753887] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[753893] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[753900] 
[753902] [MAIN] EEPROM sectors: 507, 506
[753905] [MAIN] EEPROM current: 506
[753907] 
[753909] [MAIN] EEPROM:  4096 bytes initially |  1315 bytes used (32%) |  2781 bytes free (67%)
[753917] [MAIN] Heap  : 34624 bytes initially | 20672 bytes used (59%) | 13952 bytes free (40%)
[753925] [MAIN] Stack :  4096 bytes initially |  1616 bytes used (39%) |  2480 bytes free (60%)
[753934] 
[753935] [MAIN] Boot version: 6
[753939] [MAIN] Boot mode: 1
[753940] [MAIN] Last reset reason: Reboot from web interface
[753944] 
[753946] [MAIN] Board: XIAOMI_SMART_DESK_LAMP
[753949] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ ENCODER HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[753968] [MAIN] WebUI image: LIGHT
[753970] 
[753973] [MAIN] Firmware MD5: da809ee89d9806a243bcf18ad2c37728
[753976] [MAIN] Power: 4094 mV
[753979] [MAIN] Power saving delay value: 1 ms
[753983] 

**Tools used**
* *Mac*
* *Safari*
* *Using precompiled release from [here](https://github.com/xoseperez/espurna/releases).*

Ctrl-F4, 2019-03-09:
Same experienced here.

I added
` #define RELAY_PROVIDER RELAY_PROVIDER_LIGHT`
to **code/espurna/config/hardware.h**
Then it works.
I'm new to github so don't know how to make a pr yet.

thehellodonkey, 2019-03-10:
Thanks for the tip, @Ctrl-F4 !!!
I'm new to PlatformIO and having difficulties with dependencies in the build. Any chance you could upload that binary for me somewhere while I get my mind wrapped around PIO? If not, no worries, appreciate the tip!

commit b0d47b4c (2019-03-13): Fix relay provider setting of Xiaomi Smart Desk Lamp (#1627)

fixes #1621
closed 2019-03-13 by mcspr