xoseperez/espurna #1622: How to add sensors to Nodemcu?

scarecrow2k6, 2019-03-09:
Hello everyone. I'm brand new to Espurna (coming from ESPEasy). I did some searching but couldn't find anything solid. How do I add a device (sensor)? I have a Nodemcu Lolin looking ESP8266 so I flashed the NodemcuLolin fw. I can't find any sensor section on the right pane. Please help, thanks.

I have read that I would have to manually make my own fw from the Arduino files provided. Is that true? I tried doing it but I get an error. Isn't there a way to somehow add sensors to a pre-built fw? Thanks.

scarecrow2k6, 2019-03-12:
Anyone?

ColinShorts, 2019-03-12:
If there isn't a suitable firmware that matches your use case then you'll need to roll your own firmware.

The pre-built firmware specifies which pin the sensor should be connected to. The pin can be changed by editing the code of you need to, but you will need to build the code in that event.

Read through some of the code, you'll find code/espurna/config/hardware.h most helpful. Sensors are well documented in the code and in https://github.com/xoseperez/espurna/wiki/Sensors

The are instructions in the wiki for setting up Arduino and platformio. Xose recommend platformio, and that is what I use and would recommend as it resolves the library dependencies. Try getting it working from the command line first.

If you still run into errors, try posting the error output here, or ask for help in gitter - use pastebin/similar for passing the full command and output there.

InduPrakash, 2019-03-12:
There is support for BME280/BMP280 sensors, take a look at https://github.com/xoseperez/espurna/wiki/Sensors.

scarecrow2k6, 2019-03-12:
> If there isn't a suitable firmware that matches your use case then you'll need to roll your own firmware.
>
> The pre-built firmware specifies which pin the sensor should be connected to. The pin can [...]

closed 2019-03-12 by scarecrow2k6

scarecrow2k6, 2019-03-12:
> If there isn't a suitable firmware that matches your use case then you'll need to roll your own firmware.
>
> The pre-built firmware specifies which pin the sensor should be connected to. The pin can [...]

This is super valuable information! Ok I looked into hardware.h file and see that the I2C_SDA_PIN (D2) and I2C_SCL_PIN (D1) are defined for Wemos and some other boards, but not for Nodemcu. I'm guessing it should be the same as I used the same pins with ESPEasy and it worked there. I know the BME280 requires both in general. I plugged it but still can't see any sensor section on the left hand side. What am I missing? Thanks.

ColinShorts, 2019-03-12:
Did you flash a different binary? If so, which one, what version, have you tried a shift-refresh in the browser?

There is a template for opening issues for a reason 😉

scarecrow2k6, 2019-03-12:
> Did you flash a different binary? If so, which one, what version, have you tried a shift-refresh in the browser?
> 
> There is a template for opening issues for a reason 😉

I installed "espurna-1.13.5-nodemcu-lolin.bin" for my no brand Nodemcu-Lolin (clone?). I didn't want to open an issue because I didn't know if it is an actual issue.

ColinShorts, 2019-03-12:
I don't think the lolin bin file was built with sensors built in, so you'll likely need to build your own binary.

Try setting up platformio and follow the wiki page:
https://github.com/xoseperez/espurna/wiki/PlatformIO

It's honestly the best thing you'll ever do 🙂

InduPrakash, 2019-03-12:
You can watch debug out on the serial monitor. I was once having trouble with a DHT sensor and turned out that the sensor was not connected properly due to which the sensor module was not initializing which in turn led to MAGNITUDE_NONE being returned from type() which prevented the section from being made visible in the browser. The BMX sensor will bail out if I2C initialization fails.