xoseperez/espurna #1624: 2 BMP280 on the same I2C but only one times humidity

einnordlicht, 2019-03-09:
I have succeeded to get 2 BMP280 running with the patch in  [https://github.com/xoseperez/espurna/issues/934](url). I use 2 sensors, one on 0x76 and one 0x77 by pulling SDO on Vcc. However even if both sensor support temperature, humidity and pressure measurement, only the first detected BMP280 shows humidity value, the 2nd humdity value is missing. If I exchange both sensors, again the first shows humidity, and the other not.
So in general both sensors are proven to work ok, but only humidity value of the first is shown (regardless if id 0x76 or 0x77)
Investigating the source code I realized that only for types BMX280_CHIP_BME280 the humidity was investigated (and not on BMX280_CHIP_BMP280). So first I added also this type in the checks as alternative. The I saw that on some of the assignments for humidity the index increment was missing 
(e.g. lines 88-96 of BMX280Sensors.h)  

```
            #if BMX280_TEMPERATURE > 0
                if (index == i++) return MAGNITUDE_TEMPERATURE;
            #endif
            #if BMX280_PRESSURE > 0
                if (index == i++) return MAGNITUDE_PRESSURE;
            #endif
            #if BMX280_HUMIDITY > 0
                if (_chip == BMX280_CHIP_BME280) {
                    if (index == i) return MAGNITUDE_HUMIDITY;
                  }
```
This (and other places) was changed e.g. to 
```
          #if BMX280_TEMPERATURE > 0
                if (index == i++) return MAGNITUDE_TEMPERATURE;
            #endif
            #if BMX280_PRESSURE > 0
                if (index == i++) return MAGNITUDE_PRESSURE;
            #endif
            #if BMX280_HUMIDITY > 0 
                if (index == i++) return MAGNITUDE_HUMIDITY;
            #endif
```
  and finally I got also humidity values shown for both sensors. 
However regardless what I tried I only got humidity 0 for the second detector, noting else.

einnordlicht, 2019-03-09:
Hmm, meanwhile I realized that some of the BMP280 detectors are not yielding humidity data. Replacing the sensor yield now humidity for both values, even with the original version from github

closed 2019-03-09 by einnordlicht

CraigMarkwardt, 2019-03-23:
Having two BMX280 sensors is resolved by pull #1647