xoseperez/espurna #1626: Why would an API call fail?

wsw70, 2019-03-10:
_Note: I removed the template questions as my problem is not clear-cut and I am looking for advice_

**TL;DR:** HTTP calls to `/apis` sometimes fail once, and occasionally (rarely) more than once in a row. I would like to understand why.

I added a few weeks ago monitoring for my Sonoff devices (about 20 of them). I use [`simplemonitor`](https://jamesoff.github.io/simplemonitor/) and query `http://thedevice/apis`. If I receive a `403 Not Authorized` I assume the device is OK (I wanted to do something more than a ping, to be more confident about the "higher up" functionality being live).
This check is done every minute. 

From time to time, on a more or less random device I get a failure to respond. Then, everything is back to normal on the next cycle.

The Python error for the check is

```
Requests exception while opening URL: HTTPConnectionPool(host='switch-4.swtk.info', port=80): Max retries exceeded with url: /apis (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7fd4427488d0>, 'Connection to switch-4.swtk.info timed out. (connect timeout=5)'))
```

It means that the name was resolved (so DNS is fine) and then the device did not answer to the call on time, and the call timed out.

Please note that the call was correctly "handshaked", this is not a connection which failed to establish because the port is not available or the device refuses the call. Port 80 is live but what is behind does not answer on time.

As I mentioned, this is usually a one cycle failure. I get about one or two per 24 hours of these, across the 20 devices (again, more or less randomly).

The WiFi network they are on is strong and stable (an Ubiquity Unifi AP).
Plenty of other WiFi devices (beside the 20 Sonoffs) are monitored and they never fail.
--> I would therefore be surprised if this was a WiFi problem.

Tonight the failure for one of the switches was more epic: it was not available for two hours between 3 and 5 in the morning. Everybody was asleep at that time so there is zero chance someone could have interfered in any way (and nobody has). All other services were up.

This was so unusual that I checked in the logs if indeed the checks failed ~120 times and yes, they did. The difference is that this time (and only this one) the error was 

```
/var/log/syslog.1:Mar 10 03:46:30 srv python3[24277]: 2019-03-10 03:46:30    ERROR (simplemonitor) monitor failed: switch-7 (Requests exception while opening URL: HTTPConnectionPool(host='switch-7.swtk.info', port=80): Max retries exceeded with url: /apis (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd442715c50>: Failed to establish a new connection: [Errno -2] Name or service not known',)))
```

This means that the DNS resolution failed.

I use `dnsmasq` which provides via DHCP the IP to the device and registres it in the DNS at the same time. The only reason I can think of is that the lease expired, and the device failed to renew it. And then succeeded after an hour. this may or may not be related to the other cases.

Sorry for being so detailed and thank you for having read that far.

**I wonder if there is a mechanism in the devices which would make them "go to sleep", and be awaken from that sleep by something? Or any other idea on what the issue can be?**

wsw70, 2019-03-16:
First update: I increased the timeout of my HTTP call to 11 seconds. The switches still randomly fail once, then are back one minute later on the next call. Since they are checked every minute and fail once I think that the embedded HTTP server does not always reply as expected. I will run a `tcpdump` to catch the exact traffic failing.

mcspr, 2019-03-16:
You sure devices are still on the network when it happens? What does Unifi AP logfile say?
Maybe attaching a telnet session to one of the failing devices would give some info? Something like `nc <device> 23 |& tee device.log`

wsw70, 2019-03-16:
@mcspr - good points, thanks.

**Re: WiFi** - the AP is very stable and no other device has a problem. The log shows the last disconnect two weeks ago when I did a general reboot (all the devices reconnected ~at the same time).
I do not think this is the problem.

**Re: the telnet session** - I will enable telnet on the devices and give it a try. The main problem is that I have about 20 switches and the monitoring fails randomly (both in terms of time and device). I have a `tcpdump` running at the monitoring device to record all API sessions and hope to find something once a device fails (which has not happened for a few hours, and I usually have one or two per day/coupe of days).

Now, the problem is not a blocking one - I have the impression that this is only related to the API, and only from time to time. The main drawback I have is that some of the automated updates failed for (maybe/probably) that reason. I will probably move to some OTA solution for the updates.

mcspr, 2019-03-18:
Right, tcpdump might show if esp sends some garbage. But I still got a hunch that it might self-lock itself, dropping network connection from its own POV and not reconnecting for whatever reason. Maybe having some tracing about abnormal states / disconnections would reveal that, hm.

Recent reports about Basic constantly losing connection or not keeping up softAP because we do not give SDK enough time to do WiFi stuff are my main reasoning to think that it is possible. Other boards might not react as severe, but still have the same symptoms

closed 2019-05-24 by stale[bot]