xoseperez/espurna #1628: NTP Timeout Variable in web interface

destring, 2019-03-11:
we are having issue with NTP sync and it looks like that request is getting timeout because of 1 second timeout set by default in binaries. 

I wanted to increase it but did not find any option... I think it would be handy to have this option in web interface because for some implementations where internet connection is not fast enough and remote ntp server is being used, users would want to change this...

I have also tried to set the variable using debug console but do not know which variable it actually is...  I tried to get values for "NTP_TIMEOUT" and "ntpTimeout" but I think it is not the correct variable and looks like it is hardcoded in the binaries... I know it can be changed in the source but I think making it changeable from the debug console or web interface will be helpful.

mcspr, 2019-03-25:
#1642 adds `ntpTimeout` setting
Will need to fixup some web issues before exposing it there though

mcspr, 2020-11-13:
#2132 replaces the underlying NTP client with the one from the LWIP, timeout & retries are handled via LWIP internally.
NTP_TIMEOUT / ntpTimeout are no longer used

closed 2020-11-13 by mcspr